gioblu/PJON #2: Benefits over i2c

MichMich, 2015-08-12:
Frist of: awesome work. Very impressed with the concept and final product. I'm temped to use it in my current project: 4 [automatic curtains](http://michaelteeuw.nl/tagged/curtains) controlled by one master Arduino or ESP8266. 

Before I do, I would like to know the benefits over using i2c. Is this something you can elaborate on? (Maybe even put it in the Readme doc?)

Thanks for your work. Cheers!

gioblu, 2015-08-12:
Ciao Mich I agree with you. This type of documentation and comparison will help the people to understand what is PJON. Will write as soon as possible

gioblu, 2015-08-18:
Hi Mich I studied your project a little bit and seems really well made, useful and practical! I would be happy to see PJON inside there while still in alpha! I am sorry if you see me not so active but I am in the only two weeks of vacation planned this year :) I will be back full operational in a week!

MichMich, 2015-08-19:
After some testing, I decided to go for I2C, this this is a well proven technique. [I am running into some issues](http://michaelteeuw.nl/post/127008384257/masters-slaves-and-failures), though. So I will keep PJON in mind for this project.

Enjoy your holiday!

MichMich, 2015-08-19:
BTW, did you do any testing on the maximum length of the cable connecting the two arduino's?

gioblu, 2015-08-19:
Yes works perfectly over a 30 meters long wire !

MichMich, 2015-08-19:
Then that might be a MAJOR reason to use it over I2C :)

Will give it an other try this weekend.

gioblu, 2015-08-19:
I updated my answer. Maybe you missed a part of it :)

gioblu, 2015-08-19:
In any case I would really see a well made project like yours running over PJON!! And maybe would be happy to have the same system at my home :)

MichMich, 2015-08-19:
I'll give PJON a try this weekend and write a blogpost about it on MichaelTeeuw.nl - On thing I did notice was that when I register a function with `A` as the trigger: `network.insert_reaction('A', led_blink, false);`
and I send command `B`: `network.send_short_command(44, 'B');`, led_blink will still be called. 

Maybe something to look into?

MichMich, 2015-08-19:
(if you want I can make a separate issue for this?)

gioblu, 2015-08-19:
I think is reasonable!! I am trying to understand how this bug can happen! The problem could be while storing the symbol in the reacton struct or in compare reaction function!!! I am not now able to physically test with two arduinos because I am in the only two weeks of vacation I planned this year! I will be back full operational on a week being able to contribute and fix faster!!

gioblu, 2015-08-19:
Could be a useful test to see after reaction registration if inside network.reactions[n].command_type you can find the same symbol you passed! If yes the problem should be in compare_reaction

konekko, 2015-08-21:
I would like the same thing, I am starting a project that has not even made it to pin on paper yet but I know I am going to have a shortage of pins and I need encryption and I know you offer both I am real interested in this, but need to look more into it to see if I can turn the 1 wire into a differential pair with out much work or need more info on drop per foot and effect of running along side 40VAC.

konekko, 2015-08-21:
but first enjoy your vacation :smiley_cat:

gioblu, 2015-08-21:
Hi konekko I am now using PJON over a lan wire alongside the standard house wiring inside the walls! The PJON bus is 30 meters long and passes behind fridge, microwave oven etcetera and works fine!

The drop per foot depends on the wire you are using and its length and resistance!

As soon as possible I will try to determine the drop in my wiring at different distances and its resistance!

konekko, 2015-08-25:
Perfect I’m making an alarm / home automation set up and would be nice to have minimal wiring plus use encryption for key pads and remote devices to make it harder to hack the alarm portion of it. I will keep you updated over the next 6-12 months as I work on it. May also reverse engineer PJON and cape to make them work with pi as well, will give full credit for the project idea and any code shared and let me know if there is any thing specific you want done for “PiJON” including a better name º˽º

gioblu, 2015-08-25:
Sure could be really interesting to have a port for pi! Will be a little bit tricky to time and sync but it's possible!

gioblu, 2015-08-25:
I forgot to tell you that my wiring is I star configuration so many on wires connected together in various nodes! I used a pull down resistor to kill electromagnetic spikes and reflections! Try to run the speed test and check the accuracy of the channel and add it if necessary!

gioblu, 2015-09-07:
Hi there I added in the readme the section "Why not I2c?".
Tell me if you think is satisfactory, if yes I will close the issue.

MichMich, 2015-09-07:
Perfect!

closed 2015-09-07 by MichMich