gioblu/PJON #3: Feature request: Add Multicast

MichMich, 2015-08-12:
Like /u/kloptopsmentions on [Reddit](https://www.reddit.com/r/arduino/comments/3gh5it/pjon_arduino_communication_protocol_working/), it might be a good idea to implement multicasting.

> Maybe sending to ID 0 or 254 would broadcast the message to all devices? You could use this for auto discovery of devices (like ZeroConf). Maybe even allow a way of auto assigning of ID's.

gioblu, 2015-08-12:
Ciao! To broadcast to all devices a message use BROADCAST constant as ID. Every node will receive the message but will not answer ACK  to avoid communication overlap. Thank you for your support and feature request, will read again PJON and make considerations

MichMich, 2015-08-12:
Ah, cool. Please add it to the README. 

Cheers!

closed 2015-08-12 by MichMich