gioblu/PJON #5: Documentation request: Sending arbitrary values.

MichMich, 2015-08-12:
While the documentation clearly describes on how to send short commands, it isn't completely clear how to send arbitrary values. For example, if we want to expand the blink example, by adding the blink speed to the message, how should this be done?

It would be great if a short command could contain an addition payload which would be used as a paramater to the reaction method. 

For example: 

Sender:

    void loop() {
       int blink_send = network.send_short_command(100, 'B', '200');
       delay(1000);
    }

Receiver:

    void setup() {
     network.insert_reaction('B', led_blink);
    };

    void led_blink(char *payload) {
     int speed = atoi( payload );  //char * to int conversion

     digitalWrite(13, HIGH);
     delay(speed);
     digitalWrite(13, LOW);
    };

    void loop() {
     network.receive(1000);
    };

Or is there currently a better way to do this?

gioblu, 2015-08-12:
There is send_command! Does exactly what you need! Actually the only supported data type is integer but I will for sure expand it to support other data types! I will add it to read me as soon as possible

MichMich, 2015-08-17:
Any news on this? I can't really figure out how to receive values. I'm really eager to use this in one of my projects, but unfortunately I'm a bit stuck now.

Thanks! Cheers, Michael

gioblu, 2015-08-17:
Ciao Michael. You can use the send_command function that can receive an int as third parameter:
send_command(100, 'B', 1000);

Inside the function will be called with B symbol access to network.data to read the int you have sent! 

Let me know how it goes! I

MichMich, 2015-08-18:
I already experimented a little bit with this. But in my opinion, the `network.data` is still a bit to raw. (ie. the length of the data object isn't always correct.)

In my opinion it would be better if the actual value (so, not the whole data object) was given a parameter to the function called when receiving the command.

gioblu, 2015-08-18:
Ciao Mich, thank you for your useful support! how you would implement it in an elegant way??

gioblu, 2015-08-18:
Can you please describe in which case you get an incorrect nwtwork.data length ??

MichMich, 2015-08-19:
For the implementation, I would take a look at the way it's done with I2C:

    void setup() {
        Wire.begin(4);                // join i2c bus with address #4
        Wire.onReceive(receiveEvent); // register event
    }

    void receiveEvent(int howMany) {
        while(Wire.available) {
            char c = Wire.read();    // receive byte as an char
        }
    }

About the length: The second data element tells the length of the received package. But if the IV isn't present, this package length is one byte to long. This is caused on line 183 of PJON.cpp: `uint8_t package_length = strlen(string) + 4;`

gioblu, 2015-08-22:
Hi Mich I am working on the implementation you proposed.
In the meantime I fixed the length issue you enlighted!
Thank you so much for your support!!

MichMich, 2015-08-24:
Cool thanks! I'm trying to implement it right now ... unfortunately, I'm running into a new issue: the library takes 30% of the Arduino's memory. Together with the OLED library, this won't fit. Any suggestions? (Other then buying a Teensy?)

gioblu, 2015-08-24:
Yes memory is an issue I am thinking to implement a linked list for the packets, so malloc based but I am not sure if this could provide to the user another undebuggable series of problems exceeding memory during program execution! What you think about it?

MichMich, 2015-08-24:
I think the constant is a good way to solve it. I would first focus on some of the bugs and the documentation. :)
Did you get my mail? :)

Cheers,

Michael Teeuw

	Michael Teeuw
Eigenaar Xonay Media
Xonay Media
Nieuwstraat 23B
3743 BK  Baarn	Tel: 035 887 57 07
Mobiel: 06 51 71 36 15
Email: michael@xonaymedia.nl
Web: xonaymedia.nl
Twitter: MichMich
Tumblr: michaelteeuw.nl

gioblu, 2015-08-24:
Answered you (erroneously twice)

gioblu, 2015-09-26:
Ciao!!! I added the example SendArbitraryValues  :) 
Thanks again

gioblu, 2015-09-30:
Ciao Mich! Why don't you close this issue? You can find now in examples SendArbitraryValues :)

MichMich, 2015-10-01:
Cool, thanks. I'll give it a try ASAP.

closed 2015-10-01 by MichMich