gioblu/PJON #6: Library version for IDE 1.6.5

PaoloP74, 2015-08-12:
Can you create a new branch for a new library version?
--> https://github.com/PaoloP74/PJON/tree/PJON-1.5.x

gioblu, 2015-08-12:
Hi Paolo! I would be able to maintain only one version of PJON for 0.x 1.x and 1.6x!
I would merge your contribution but I am not sure if I want to have digital write fast and PJON in the same arc directory! And also I can t get the sense of having the src directory! Generally is used in repos where u can / have to compile or maybe I am ignorant :)

gioblu, 2015-08-12:
In any case I really thank you for your effort! i will for sure include all other necessary mods u did

PaoloP74, 2015-09-06:
OK, i add a PR whit only library.properties and kaywords files.

gioblu, 2015-09-07:
Ciao Paolo, thank you so much for your contribution. I discovered with your suggestions on the arduino forum how the new library structure is made after 1.5 version, so as you correctly did in the closed pull request, I should create a 1.5x compatible master version with that modifications. I will soon set is as master, if you want to contribute you will be able to pull request the new features there.

closed 2015-09-21 by PaoloP74