gioblu/PJON #12: Document the standard

dnet, 2015-09-27:
I see that the source code has plenty of comments, but a specification be much more readable in a separate Markdown file or (GitHub) wiki page.

(continued from https://news.ycombinator.com/item?id=10283322)

gioblu, 2015-09-28:
Hi, thank you for opening this issue. I started working on the PJON wiki, please, consider it as a "public draft" with a lot of work in progress. Do you have some suggestion on how to structurize this wiki, if you think is necessary to divide it in pages and if you like ascii schematics or you would prefer images??

Thanks you again

dnet, 2015-09-28:
Thanks, it looks nice. I'm good with ASCII art, and will take a more detailed look at it soon.

gioblu, 2015-09-30:
Ciao dnet What do you think of the actual state of development (early) of the wiki?
https://github.com/gioblu/PJON/wiki

You perceive it easy to understand?
You think I missed something or the format is not coherent with a standard reference?

dnet, 2015-10-02:
Looks good!

gioblu, 2015-11-24:
Hi dnet, now that the standard is documented, why don't you close the issue? Or you think documentation has some downsides?

closed 2015-12-21 by gioblu