gioblu/PJON #13: ATtiny compatibility

dnet, 2015-09-27:
I see that it uses Wiring, but I'd be interested what the minimum requirements are for a compatible implementation. For example, would it be possible to implement it without some serious bit hacking on an ATtiny?

(continued from https://news.ycombinator.com/item?id=10283322)

q2dg, 2015-09-28:
Maybe this comment should be placed in another issue but...is it possible ARM (Due & Zero) compatibility? Thanks

gioblu, 2015-09-29:
Ciao, ARM is still not compatible. Why don't you open a dedicated issue for that?

gioblu, 2015-09-30:
Ciao dnet. I digged a littlebit on the web and I think it is possible that this library runs without modifications on ATtiny85 16Mhz (external and maybe also with internal PPL clock). It is necessary to select a custom porting for ATtiny on Arduino 1.6 IDE to have Wiring and be able to code as you normally do with Arduino boards. If you want to give it a try I can support you.

gioblu, 2015-12-28:
ATtiny45/85 compatibility added:
https://github.com/gioblu/PJON/commit/bdc0b570e2d017b56cd633dc48972805f9fc9e41

As you can see I reduced the timing value of BIT and BIT_SPACER to match the other boards produced duration and added to digitalWriteFast PJON version the ATtiny compatibility.

I will close the issue for now, reopen if not satisfied or need other ATtiny processors.
Marry Christmas!

closed 2015-12-28 by gioblu