gioblu/PJON #14: Raspberry PI compatibility

gioblu, 2015-09-29:
Ciao! I am considering which is the best and most efficient approach to implement the PJON standard on Raspberry. Do you have some suggestion? I sincerely would remain on C / C++ / Wiring.

gioblu, 2016-09-14:
See https://github.com/Girgitt/PJON-python 
Try PJON on python running on a raspberry :)

gioblu, 2017-03-05:
It seems possible to port PJON to raspberry, so I reopen this issue after some time of inactivity; tt is not sure all strategies will be able to perform as on the other compatible devices, but the most common, as Serial, should work fine. Firstly some portability changes have to be made (i.e. boolean type switch to bool, ecc.). I am evaluating the possibility of implementing a group of macros pointing to the methods used for IO, timing, random and Serial around the PJON codebase. So it could be easy to port a totally different architecture / device compiling this macro list pointing to the correct methods.
```cpp  
- interfaces
 - ArduinoInterface.h
 - RaspBerryPiInterface.h
```
An example of interface (RPI): 
```cpp  
#if defined(RPI)
  #include <inttypes.h>
  #include <stdio.h>
  #include <stdint.h>
  #include <string.h>
  #include <WiringPi.h>

  /* Fallback to WiringPi core functions ------------------------------------ */

  #if !defined(PJON_IO_WRITE)
    #define PJON_IO_WRITE digitalWrite
  #endif

  #if !defined(PJON_IO_READ)
    #define PJON_IO_READ digitalRead
  #endif

  #if !defined(PJON_IO_MODE)
    #define PJON_IO_MODE pinMode
  #endif

  #if !defined(PJON_IO_PULL_DOWN
    #define PJON_IO_PULL_DOWN(P) { \
      PJON_IO_MODE(p, INPUT); \
      pullUpDnControl(P, PUD_DOWN); \
    }
  #endif

  /* Random ----------------------------------------------------------------- */

  #ifndef CLDL_RANDOM
    #define CLDL_RANDOM random(randMax) (int)((1.0 + randMax) * rand() / ( RAND_MAX + 1.0 ) );
    /* Scale rand()'s return value against RAND_MAX using doubles instead of
       a pure modulus to have a more distributed result */
  #endif

  #ifndef PJON_RANDOM_SEED
    #define PJON_RANDOM_SEED srand
  #endif

  /* Serial ----------------------------------------------------------------- */

  #ifndef PJON_SERIAL_AVAILABLE
    #define PJON_SERIAL_AVAILABLE serialDataAvail
  #endif

  #ifndef PJON_SERIAL_WRITE
    #define PJON_SERIAL_WRITE serialPutchar
  #endif

  #ifndef PJON_SERIAL_READ
    #define PJON_SERIAL_READ serialGetchar
  #endif

  #ifndef PJON_SERIAL_FLUSH
    #define PJON_SERIAL_FLUSH serialFlush
  #endif

  /* Timing ----------------------------------------------------------------- */

  #ifndef PJON_DELAY_MICROSECONDS
    #define PJON_DELAY_MICROSECONDS delayMicroseconds
  #endif

  #ifndef PJON_MICROS
    #define PJON_MICROS micros
  #endif
#endif
```
So a capitalized macro `PJON_MICROS` substitutes micros in PJON codebase, without performance loss, thanks to the preprocessor.

In PJON.h will be something like:
```cpp
...
uint16_t receive(uint32_t duration) {
uint16_t response;
uint32_t time = PJON_MICROS();
while((uint32_t)(PJON_MICROS() - time) <= duration) {
...
```

It is a change that would affect most of the project, so it should be evaluated carefully. By my side it seems could really enhance portability (not only of PI, but organize a way to port various other systems), maybe with a slight loss in code readability. On the other side a ton of #ifdef(RPI) in the implementation seems for me less elegant and mantainable (although few will be necessary).

@fredilarsen @ekarlso @4ib3r what do you think about it? :)

(in the code example is missing analogRead, working on it)

fredilarsen, 2017-03-05:
As it increases portability at the cost of a little readability but without any increase in storage or RAM, or decrease in speed, I say go for it.

Including the Interfaces.h header at the start of PJONDefines.h will make sure the defines are available for all strategies included in PJON.h as well. So I cannot see a need for restructuring, just a manual search-and-replace to use the new defines.

gioblu, 2017-03-10:
CIao, I added the initial sketch of the interfaces, it enable raspberry pi to compile PJON and run ThroughSerial strategy, see: https://github.com/gioblu/PJON/wiki/Raspberry-Pi-interfacing

Is there someone would support us with some testing?

closed 2017-04-16 by gioblu