gioblu/PJON #15: ARM compatibility

q2dg, 2015-09-29:
It would be terrific if this library could run on Due and Zero boards.
Thanks

gioblu, 2015-09-30:
Thank you for your suggestion. As soon as I wil finish documentation and wiki I will buy a Due and will try to make some experiments. Do you have a Due?

q2dg, 2015-10-01:
No, sorry :-(

gioblu, 2015-10-08:
Thanks to Esben we know that on Arduino Zero to Arduino duemilanove is not working. From his oscilloscope's reading it seems that the spikes are smaller and less precise. In my opinion Arduino Zero microsecond duration is different, probably shorter then Arduino duemilanove.
I will soon try out Arduino Zero and will try to set BIT_WIDTH to a different value, to match the real microsecond duration of Duemilanove. Will let you know results.

Earlz, 2016-01-08:
Have you tried this on any other ARM processors, such as the LPC1768 (ie, mbed)? 

Does it come down to instruction timing and/or does some of this need to be written in assembly to fit within a certain timing profile ?

gioblu, 2016-01-09:
Hi there, I never tried mbed, but I will experiment with it as soon as I will be able to port ESP8266.

I don't think it needs assembly code or convulted solutions, but I am not sure which could be the right one in a multithread environment, I am still studying this architectures to have a more clear idea.

Earlz, 2016-01-10:
I have an mbed and should have an oscillioscope sometime soon, so hopefully I can contribute.. This is definitely not my expertise though so no promises

Earlz, 2016-01-10:
I got it to compile for mbed using the web compiler and I think things are correct.. but I can't get any communication to work yet. If you want to take a peek, the code is at https://developer.mbed.org/users/earlz/code/pjon_testing/ (will move it to github once it's actually working)

gioblu, 2016-01-10:
Thank you so much, I am reading your code, will give you feedback soon.

Earlz, 2016-01-13:
I took a look at all of this with my newly acquired oscilloscope and found out that the pin I was testing PJON with on my mbed is apparently bad... but changing the pin to something else also didn't immediately fix it. I did some simple timing tests as well by just generating a square wave with `delayMicroseconds(X); digitalWrite(0, Y)...` Down to 200us, both the ATTiny85@8MHz/3.3v and the mbed running at 96MHz/3.3v seem to agree on the speed and I can't see any real difference between the waves with my scope. However, using a 50us period for the square wave, the mbed is probably about 10-20% slower, which is a bit unexpected to me. This may be because of the ATTiny's internal clock jitter. I haven't tried using an external clock yet. And I'm still figuring out how to measure the actual wave's period with my scope with certainty, so thus far I can only really give relative read outs.

gioblu, 2016-01-13:
Hi Earlz, Really interesting feedback. I am sure I will be able to suggest you a solution with these readings. 

Read carefully these pages in the wiki:
https://github.com/gioblu/PJON/wiki/ATtiny-interfacing
https://github.com/gioblu/PJON/wiki/Troubleshooting

Consider that execution time is probably different in these architectures, consider ATTiny's as the canonical speed and try to adapt the other. Consider that READ_DELAY value is extremely important to get a stable communication because is the value that syncronize the two devices, prepending some time before bit reading to match the other architecture's execution and time bias. So try to change it running the network testing examples. If you get also few damaged packets per second it means that you are really near to success.

gioblu, 2016-01-13:
Another interesting hint was made by lucky, in Agenda another library I am developing. In any case, he, working to port Agenda to ESP found a huge bug because of the fact that types in a 32 bit system are differently memory allocated see:
https://github.com/gioblu/Agenda/issues/1

As I fixed in Agenda, probably also PJON needs typecasting or temporary variable where to execute calculation where type is not brought explicit by code to avoid 32 bit standard type memory allocation bug.

Earlz, 2016-01-16:
I tried some more stuff for timing, but I feel like this might not actually be able to tolerate so much potential inaccuracy as the ATTiny85's internal oscillator. I got it to receive a few "busy" packets, but then it proceeded to freeze up in some way that I don't understand. (ie, a "this should never happen" type thing). I'm still working to see if I can get it to properly handle packets

 I feel like to properly do this however there needs to be a timing calibration phase where each device figures out it's own relative inaccuracy in clockrate. Also, I don't see any way that this protocol avoids two masters attempting to send conflicting messages at the same time

gioblu, 2016-01-16:
Please let me see your code, I will try to identify where could be the problem.
If it is freezing up is probably because of a buffer oveflow and / or for type compatibility as Lucky enlighted. As many other protocols, is not completely collision free, but there is a really slight chance that 2 devices can speak in the same time. If this happens, message is detected as broken and retrasmitted.

Earlz, 2016-01-16:
I published my changes at https://developer.mbed.org/users/earlz/code/pjon_testing/ However, I don't even think it's a buffer overflow, or at least not an obvious one. it seems to at some point stop updating the timer is my best guess, but I don't have any real debugging tools for this other than toggling some LEDs so it's pretty difficult to figure out. The problem is somewhere in send_string, pjon.cpp line 227. in the receive loop. (it never exits the loop)

gioblu, 2016-01-16:
Thank you for your support, and I am sorry because I perfectly understand your anger / frustration :).I am taking a look to the code, I hope to find something. Will soon give you some feedback.

In any case, I did a little tweaking to the values. I did a test of all the supported devices to be sure there was nothing hidden that could have blocked you. I found only a slight reliability loss because of some recent modifications I did to timing comparisons to avoid micros() overflow bug. I had only to tweak slightly to get back the 10% was lost. All seems fine on all supported architectures in all the possible combinations (I needed all today to do that).

In any case, a pull is suggested to have the actual master.

gioblu, 2016-01-16:
From a fast and initial analysis seems fine. For a faster debugging I suggest you to test the all low level functions in an isolated way, for example try with really slow speeds to call network.send_bit() and visually check with a LED, network.send_byte() and see if the LED flashes have corrispondence with the bits of the character you passed as parameter, After this try send_string() and see if the flashes are ok, and so on, start from low level and check all function separately going up.

gioblu, 2016-01-16:
Using this method you will probably find in which function it freezes

adrianrolla, 2016-11-23:
Gioblu, I have a DUE, uno, nano, mega and some time to help with. 
Tryed to make it work with no success, I just could make a DUE send a byte to UNO/MEGA with ACK ok, but could not get UNO/MEGA send anything to DUE.
If you wish, let me know what can I do in order to try making DUE work with pjon. Regards!

gioblu, 2016-11-23:
Ciao @adrianrolla, using `ThroughSerial` should work, but if you are interested in `SoftwareBitBang`, you should find the constant definining your architecture / processor (see https://github.com/gioblu/PJON/blob/master/strategies/SoftwareBitBang/Timing.h) write a new section with the timing definition related to Due and try to tweak its timing, changing `SWBB_READ_DELAY` (could also be negative). Or if no success, try changing `SWBB_BIT_SPACER` and `SWBB_BIT_WIDTH` and `SWBB_ACCEPTANCE` accordingly. Seems complicated, but it could need 10 minutes to find the optimal values. 

Being Due multitasking architecture, could have issues to receive bytes because interrupted by other processes, thing that should not happen using `ThroughSerial`.

gioblu, 2016-11-23:
@adrianrolla see also https://github.com/gioblu/PJON/wiki/Troubleshooting

gioblu, 2016-11-23:
@adrianrolla be sure to be on master

adrianrolla, 2016-11-23:
Ciao @gioblu , many thanks for your answer.
Will do both things `ThroughSerial` and try to tweek a new section in Timing.h for `SoftwareBitBang`
Whatever the outcome is,  will let you know about the results.
Regards!

gioblu, 2016-11-23:
@adrianrolla feel free to write if needed!

gioblu, 2016-11-24:
A solution if no success is obtained with SoftwareBitBang on Due could be to use a ATmega family router from ThroughSerial to SoftwareBitBang (So running two different PJON instances with the described strategies)

gioblu, 2018-07-16:
Closing because duped, see or contribute in the more recent one.

closed 2018-07-16 by gioblu