gioblu/PJON #19: Not working with mega 2560

morpheus000, 2015-12-28:
hi there, i cant get a network up and running with two uno's and one mega2560, if i use a third uno PJON works like a charm.
Maybe i found the right excuse to buy a new oscilloscope!

gioblu, 2015-12-29:
Ciao! It's a pleasure to hear you tinkering with PJON. 
Please explain in a more detailed way your setup.

Please be sure to have the actual master version.

gioblu, 2015-12-29:
I will post you a fast checklist to be sure, follow this for every device you wire:
1. Check wiring
2. Check pin number assignment in code
3. Be sure that common ground is present
4. If you are working with a mixed device network be sure to run PJON in COMPATIBILITY_MODE
(in PJON.h line 57 COMPATIBILITY_MODE should be true) as it is by default.

If you have communication problems try to run network_analysis example with Uno/Duemilanove/Nano and Mega (in both directions) and if you can post here the results.

morpheus000, 2015-12-29:
I already checked the 4 points you mentioned but your network_analysis example didn't crawl in my tired brain yesterday :) will give it a try later today and post the results as well as the setup!
An older version (circa 2 month old) is working though there is some data loss.
Thanks for the great work, i going to use pjon as short-range protocol for my home automation.

morpheus000, 2015-12-29:
Seems like i have to be sorry for opening that issue, i can not recreate the error!
Its working now, the only thing i got often are messages send twice, but thats not an issue.
Just as explanation, the mega is forwarding messages incoming from ethernet/mqtt to your pjon protocol. I don't know what i changed but it works, i will try to recreate my mistake.

gioblu, 2015-12-29:
Ciao Morpheus! Not a problem! I am happy you have it up and running!
Why don;t you send me some pictures and a short description of your project? I will add it to the application examples!

I will close the issue
Marry Christmas

closed 2015-12-29 by gioblu

gioblu, 2015-12-29:
Ciao Morpheus, the double sending is a known issue related to the COMPATIBILITY_MODE stadard timing setup. I am working on it and will fix before vacation finishes

morpheus000, 2015-12-29:
I will gladly send you something if there is a fix concept.
PJON is already a really brilliant work, keep it up!
I'm going to stress test your protocol soon under really tough conditions, unshielded wiring installed beside the 230V wires for at least 30meter.... really curious about the result.

gioblu, 2015-12-29:
Happy to hear about it! Good luck I am doing the same at home here with cat 5 lan wires and goes perfectly. If you have some problem / issue / bug reports, open an issue!

morpheus000, 2015-12-30:
Ciao globlu, to you sometimes get messages mixed around( send messages in order 1 2 3 receive messages in order 1 3 2) too or is ah problem related to my code?
Thanks for being so helpful!

gioblu, 2015-12-31:
Ciao Morpheus please post here the code. Thank you very much again.

gioblu, 2015-12-31:
Ciao Morpheus, so I made a little analysis. I think your setup is mega to mega in compatibility false.
In this setup mega to mega has not reliable communication because the actual duemilanove/uno standard timing values are too short for mega, and so some times some packets are received not correctly or lost. I have only to tweek to find a common setup for duemilanove and mega a littlebit higher than 20 microseconds. Will fix in the next days.

In any case PJON doesn't ensure packet reception in the order was inserted in the sending queue.
If you need for some reason an indexed packet you can add it by yourself inside the packet, as part of the data. I think, considering the complexity to send in order all the packets for all the devices, it's not efficient to add this functionality at so low level, slowing down many use cases where order in not required.

gioblu, 2015-12-31:
In any case happy new year. Pull PJON in the next days to see solved this little issue.

morpheus000, 2016-01-01:
Ciao gioblu, my setup is mega to uno/nano.
If packets are mixed around its not a problem for my application of pjon, i just wanted to mention it in case you were not aware of this.
Happy new year too and thanks again!

PS: I will notify you when i have tested my setup again with your new code.

gioblu, 2016-01-01:
Happy new year to you too! Good luck for your application, email me here gioscarab@gmail.com if you would like to share your build in the new section "Application exampes" I will add in the wiki soon.