gioblu/PJON #20: Latency amount?

Earlz, 2016-01-02:
I didn't see any obvious answer, but what is the expected latency of this protocol and maximum latency? Going just from the speed of compatibility mode, I get 1.8ms to send a byte.. but I'm pretty sure that's not the right way to figure out latency, especially maximum latency with error correction and such

gioblu, 2016-01-03:
Byte transmission time:
- 236 microseconds in standard mode
- 472 microseconds in compatibility mode.

Packet latency depends on the **length of the packet**, **bandwidth occupation by other devices** and **noise**. Expected latency can be calculated using data throughput:

packets/s = 2.6kB/s or 1.3kB/s  / (packet length + 3) (internal symbols: ID, length, CRC)

Standard mode:
How many custom sized packet / s = 2600 / 4 (a packet of a single byte)
Expected latency = 1000 / custom sized packet/s = 1.49 milliseconds
671 packets / second
 
Compatibility mode:
custom sized packet / s = 1300 / 4 (a packet of a single byte)
Expected latency = 1000 / custom sized packet/s = 3 milliseconds
333 packets / second

Expected latency as shown above, is more precisely called round trip delay time, because also ACKnowledge response time is included in data throughput.

gioblu, 2016-01-03:
Probably this should be added. explained more clearly, in the readme

closed 2016-01-09 by gioblu