gioblu/PJON #22: Is it possible to get the sender id?

TeeTrizZz, 2016-01-08:
Hi gioblu,

very nice library! I have a question, though. Is there a possibility to get the sender id of a package? I saw that the set_receiver function's comment had a param for this in an older version. I guess this has never been implemented, has it? Should I send the sender id as part of the content?

Best,
Fabian

gioblu, 2016-01-08:
Hi Fabian, I was near to implement it but after some calculations I decided not to insert sender_id in the standard packet header. This because it generally slow down every communication and not every one needs to transport this information in the packet, for this reason I chose to let the user decide if to send it or not. 

What you think about it?

TeeTrizZz, 2016-01-09:
Thanks for the fast reply! I already thought that this was the reason. It's no problem, though. I'm going to transmit the sender_id as the first byte of the content. I just wanted to be sure that this is not implemented :-)

gioblu, 2016-01-09:
Good luck for your project!
Contact me if you want to add your build to the PJON's application examples in the wiki.

gioblu, 2016-01-09:
I will close the issue. Have a nice weekend

aperepel, 2016-03-10:
Hey guys, decided to check any closed issues before I file one, and, bingo!

Is it possible to control the sender I'd inclusion as a configuration switch for the library? Both arguments above are valid. Not sure if you can incorporate it into an existing protocol or maybe it's time to have protocol versions? ;)

gioblu, 2016-03-10:
Ciao @aperepel, thank you for your support and for sharing your opinion.

PJON Standard is designed to offer the minimum you need for a one-to-many or many-to-many communication in multimaster configuration, defining a basic bus line where the end user can implement through his sketches its own implementation needs, for example adding the sender id as the first character of the content. I think is better in terms of efficiency to let the user implement his own layer of abstraction over PJON than trying to cover all the cases with it (see low level internet stack protocols).

For this reason in version 1.1 was added the getter method `get_id()`

In my opinion configurability is not always a good choice, for this reason I am working on this standard to offer an almost zero-configuration system that, I think, will ease maker's life in the new embedded systems network scenario of the next years :)

If you go back with commits you will see that PJON in the older versions and at the initial stage of its development was completely configurable. That gave the chance to me and other users to experiment its limits and select the "operational envelope", and consequently remove configurability.

But, as it should be, more and more PJON is developed by the community rather than only by me, so I would be happy to know your, and others opinion to find the best solution :)

gioblu, 2016-03-10:
@aperepel for sure this is a lack in the master documentation. Should be added an example where is explained how to add the sender id as first character of the payload. Please don't think I was triyng to advertise my opinion. 

My only concerns for this issue are:
- Consistent overhead adding also the sender_id in any case, for every packet, 
- Speed loss because of longer function execution time (more parameters, ecc.), for every packet
- The amount of added complexity and potential bugs (in the library and in the end user's code)
- Inconsistencies between architectures this addon can lead to

Let's consider the overhead amount as is now:
- 1 byte reading the channel to be sure not to create a collision
- 1 byte for the recipient id
- 1 byte for the length of the packet
- 1 byte for the error detection
So 5 bytes to send 1 or 15 to send 10 and so on...

Lets consider what is better to do, I will leave in any case this issue opened to receive feedback from other users.

aperepel, 2016-03-10:
Hi, I agree the overhead might be too much if we are going 'micro'. For basic cases where there's 1:1 comms only, it may not make much sense even. So, if there's a quick doc example people can reference (read, copy/paste), this should address things without the need to add it to the protocol.

littleprojects, 2016-04-05:
I think, this would be an important feature. If you get the sender id, you can easy make a pjon scanner. Or you can use the ids not for the device idetification but for the content identification. 
You can use the id for device or content or both. what you need. it makes its more flexible. 
This is a realy awesome work!

gioblu, 2016-04-05:
Ciao @littleprojects. Thank you for your compliments and support. One of the IDs is dedicated to `BROADCAST` to all devices. Every device will receive the message but will not respond with ACK. This gives you a slight performance gain but you loose the certainty of packet delivery. Maybe some of them are busy executing something and will not receive the message and you are not aware of it, so it's useful for some key situations like general assert to all devices repeated every 5 seconds where you don't need reception certainty.

I wil take a second to reason on your proposal of content_id to have a proper answer to you.

littleprojects, 2016-04-05:
If we can use the sender_id as a content_id, we can save 1 byte for adressing. 
When i send 1 byte i need 40 bits. When i need a extra byte for a content_id in the payload i have 48 bits. With a sender_id as content_id i can reduce the trafic about 20%. We can say, it 20% quicker. 

So i combine the posetives of I2C / OneWire and CAN if i use the message Id as a sender_id or content_id. And its is little bit quicker ;)

gioblu, 2016-04-06:
Ciao @littleprojects.
Being PJON a non-interrupt driven system and because of the structural choices of the Standard itself, would not be possible for the transmitter to be sure that all devices expecting a particular content_id received the message (ACK is syncronous). For this reason considering this loss, I don't see sparing saving a byte to be as decisive as necessary to add the expected complexity to implement what you are describing. But there is always the chance I misunderstood or didn't get your point :)

gioblu, 2016-04-06:
We could bring back the old setter from 0.x that was enabling/disabling `ACK` transmisssion and reception. If loss of acknowledgment is a choice of the user, he/she could call `network.set_acknowledge(false)` and then use the ID in the way he prefer.

closed 2016-04-13 by gioblu