gioblu/PJON #23: How to get working with ATTiny85?

Earlz, 2016-01-10:
I'm trying to get this working with an ATTiny85 and having problems with the Arduino IDE. I installed the ATTiny package for the Arduino IDE, but get these compiler errors:

<pre>

Build options changed, rebuilding all
In file included from C:\Users\Jordan\Documents\Arduino\libraries\PJON\PJON.h:33:0,

                 from C:\Users\Jordan\Documents\Arduino\libraries\PJON\PJON.cpp:23:

C:\Users\Jordan\Documents\Arduino\libraries\PJON\PJON.cpp: In member function 'boolean PJON::can_start()':

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:92:28: error: 'DDRD' was not declared in this scope

 (((P) >= 0 && (P) <= 7) ? &DDRD : (((P) >= 8 && (P) <= 13) ? &DDRB : &DDRC))

                            ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:125:12: note: in definition of macro '__atomicWrite__'

 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \

            ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:144:88: note: in expansion of macro 'digitalPinToDDRReg'

 do {if (__builtin_constant_p(P) && __builtin_constant_p(V)) __atomicWrite__((uint8_t*) digitalPinToDDRReg(P),P,V) \

                                                                                        ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\PJON.cpp:82:3: note: in expansion of macro 'pinModeFast'

   pinModeFast(_input_pin, INPUT);

   ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:92:71: error: 'DDRC' was not declared in this scope

 (((P) >= 0 && (P) <= 7) ? &DDRD : (((P) >= 8 && (P) <= 13) ? &DDRB : &DDRC))

                                                                       ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:125:12: note: in definition of macro '__atomicWrite__'

 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \

            ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:144:88: note: in expansion of macro 'digitalPinToDDRReg'

 do {if (__builtin_constant_p(P) && __builtin_constant_p(V)) __atomicWrite__((uint8_t*) digitalPinToDDRReg(P),P,V) \

                                                                                        ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\PJON.cpp:82:3: note: in expansion of macro 'pinModeFast'

   pinModeFast(_input_pin, INPUT);

   ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\PJON.cpp: In member function 'void PJON::send_bit(uint8_t, int)':

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:90:28: error: 'PORTD' was not declared in this scope

 (((P) >= 0 && (P) <= 7) ? &PORTD : (((P) >= 8 && (P) <= 13) ? &PORTB : &PORTC))

                            ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:125:12: note: in definition of macro '__atomicWrite__'

 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \
[... 642 lines cut ...]

                                                                         ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:125:12: note: in definition of macro '__atomicWrite__'

 if ( (int)(A) < 0x40) { bitWrite(*(A), __digitalPinToBit(P), (V) );}  \

            ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\includes/digitalWriteFast.h:137:86: note: in expansion of macro 'digitalPinToPortReg'

 if (__builtin_constant_p(P) && __builtin_constant_p(V))   __atomicWrite__((uint8_t*) digitalPinToPortReg(P),P,V) \

                                                                                      ^

C:\Users\Jordan\Documents\Arduino\libraries\PJON\PJON.cpp:351:7: note: in expansion of macro 'digitalWriteFast'

       digitalWriteFast(_input_pin, LOW);

       ^

exit status 1
Error compiling.

</code>

Earlz, 2016-01-10:
Er, don't use the version from the Arduino library service...

closed 2016-01-10 by Earlz

gioblu, 2016-01-10:
What you mean! I didnt get this issue!