gioblu/PJON #25: Digispark ATtiny85 usb board compatibility

gioblu, 2016-01-23:
Test and if necessary add compatibility to Digitspark ATtiny85 usb board.

aperepel, 2016-03-17:
I have the DigiSpark as well as some generic attiny85 boards. Wanted to help out for a while, tied up with something else currently. If it's nothing urgent, feel free to assign to me, please.

gioblu, 2016-03-17:
It's a pleasure to get some support to this issue. Thank you @aperepel 
I will in this case assign to you the issue! Please let me know your results with Digispark.

aperepel, 2016-03-20:
I've spent some time on it, but couldn't make Attiny85-based dev boards to work. My bare-bone attiny85 chips haven't yet arrived, so I can't attest to results mentioned here: https://github.com/gioblu/PJON/wiki/ATtiny-interfacing Would like to know more detail from whoever succeeded in those tests.

Trying out with all 6 available pins produces only Failed reports, with P3 reporting Busy. This busy signal is for a different reason, though. My understanding is they have a 1.5KOhm pull-up resistor on it and it's also used for USB when connected. Anyway, it produced Busy regardless of whether PJON network was configured to use this pin, so I'm inclined to ignore it.

It could be something to do with timing, but I don't have an oscilloscope to confirm (nor any experience to use it :) ). Maybe I could try again once my bare-bone AtTiny85 chips finally arrive.

aperepel, 2016-03-20:
As a side-note, the esp8266 to esp8266 communication worked with a 10K pull-down resistor. With AtTiny85 and Uno/Esp8266/AtMega32u4 I didn't get any busy signal at all. Same with 1MOhm.

gioblu, 2016-03-20:
Hi @aperepel I personally had success with a raw chip and a generic breakout board. Generally pull-down resistor was not necessary to have correct data communication. Probably, Digispark toolchain use different compilation optimization flags, and so function execution time could variate compared to the attiny porting of Mellis (that is the one suggested in the wiki and the one used for testing).

Actual timing values are optimized for Mellis toolchain and 8Mhz internal oscillator, probably should be found a way to identificate the Digispark toolchain, and wap the correct timing for that in a conditional as was done for all the other architectures.

aperepel, 2016-03-20:
IIRC, Digispark and all dev boards for AtTiny85 are 16Mhz, that would explain it.

budaics, 2016-12-26:
Can you help me whats going on?

[image]

gioblu, 2016-12-27:
Ciao @budaics, thanks for the report. Ethernet is not present in the ATtiny core so it complains.
Will soon push and relate a commit to this issue with the fix. If you need it now, simply comment the lines 78 and 79 (https://github.com/gioblu/PJON/blob/master/PJON.h#L78) in PJON.h
Thanks again.

budaics, 2016-12-27:
Ciao,

I just found and I really love this library, much more easy than the others. Commented those lines and the build was successful. Thank You for the help. But sorry for saying that, Attiny85 doesn't work for me.

On the left there is an arduino/transmitter.
On the right with attiny85 not working, receiver_function not called.
If i change right to arduino, everithing is ok again

[image]

gioblu, 2016-12-27:
Ciao @budaics if your device includes a dedicated external oscillator the motivation could be because there is not dedicated timing for that clock. See strategies/SoftwareBItBang/Timing.h. If there are no compatible timing, you can simply try to define them yourself as many users did.

If your ATtiny85 is running its internal oscillator at 8Mhz, experiments have shown this component to be extremely unreliable and behaving differently chip to chip, so you can find some ATtinies working perfectly, others not working at all, due to clock differences. More stable oscillators gives a more reliable performance (see Arduino Duemilanove / Uno, of different distributors, with different hardware components, made in different countries, all perfectly interoperable at 4.87kB/s in OVERDRIVE mode).

Sadly this is not the case and an external oscillator (with its dedicated timing) seems to be mandatory to have PJON running nominally on ATtiny85, always :). This necessity steals 2 pins, and most of use cases (also in my experience) are cut out because of the lack of pins, for this reason, few (and also me :) ) went after experimentation test and didn't support the development of this porting.

Thank you for your report, and feel free to ask my and other users support to bring forward this side, thanks again.

budaics, 2016-12-27:
Thank you again, 16 MHz external oscillator solved the problem! :)

gioblu, 2016-12-27:
Hei @budaics happy to hear you fixed, why don't you share your setup here or make a pull request so other users with external 16 will be able to follow your steps :) ?

gioblu, 2016-12-28:
Ciao @budaics, thanks a lot, I fixed few minor things in the wiki, seems the default values are ok for ATtiny85 at 16Mhz external! Cool! Maybe they should be defined in the condition related to ATtiny to have them dedicated. Could you run the networkAnalysis example and print me here the results?

budaics, 2016-12-28:
```
---------------------
Packet Overhead: 6B - Total: 414B
Maximum Bandwidth: 1794B/s
Data throughput: 1380B/s
Packets sent: 69
Mistakes (error found with CRC): 0
Fail (no answer from receiver): 0
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
---------------------
```

gioblu, 2016-12-29:
Ciao @budaics, thank you for your tests, fantastic it seems working perfectly with the default timing!
It may work also in mode `FAST` simply by default!

```cpp
 /* Set SoftwareBitBang mode to _SWBB_FAST before PJON.h inclusion
     (Transfer speed: 25.157kBd or 3.15kB/s) */
  #define SWBB_MODE 2

  #include <PJON.h>
```

budaics, 2016-12-29:
With ` #define SWBB_MODE 2` mode
```
---------------------
Packet Overhead: 6B - Total: 582B
Maximum Bandwidth: 2522B/s
Data throughput: 1940B/s
Packets sent: 97
Mistakes (error found with CRC): 0
Fail (no answer from receiver): 0
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
---------------------
```

gioblu, 2016-12-29:
Ciao @budaics cool, thank you again for the great news! So you assessed nominal functionality on both `STANDARD` and `FAST` modes without the necessity of dedicated timing.

gioblu, 2016-12-30:
Ciao @budaics here the fix to the inclusion error you reported: https://github.com/gioblu/PJON/commit/a2561bd36338b128559e360bf8b8390ca61807aa

thanks again.

Aypac, 2017-06-04:
It appears to me, that this issue is resolved. Is that true?

gioblu, 2017-06-04:
Ciao @Aypac, ATtiny85 used with a 16Mhz external oscillator works fine with SoftwareBitBang strategy! See https://github.com/gioblu/PJON/wiki/ATtiny-interfacing

I still didnt have the time to try with the digispark usb version, although seems less interesting and feasible because of the pins and program memory loss due to the usb port / bootloader.

Aypac, 2017-06-04:
If you are not actively using the USB port, you won't lose any ports in the digispark board. And in many applications the memory loss is not really an issue.
At the same time it provides a very nice platform for Prototyping and playing around, because everything is there already. 

The Quartz on the other hand takes 2 of the 6 ports, leaving only 3 for non-PJON uses.

So is there a possibility to reduce clock speed or something to not having to use a quartz? I mean even a clock line would take fewer pins in this case and might thus be beneficial.

gioblu, 2017-06-07:
Ciao @Aypac I have tried personally using the internal PPL clock, with scarse results, it is too unreliable and its behavior changes chip to chip :(. The clock line is a possible solution to the "not enough pins" problem and I agree could beneficial although PJON is designed to be as general as possible and so I do not see the addition of a clock line as something feasible in all the possible use cases. It could be an optional feature activated through constants, inside the SoftwareBitBang strategy. I sincerely do not have an enormous experience bringing the clock line around, is that reliable on "long" distances  or in the same operational envelop of SoftwareBitBang (for now tested up to 50m)?

dontsovcmc, 2017-11-21:
Today I got Digispark (Attiny85 16Mhz) connected to Arduino Micro (32u4) as described here:
https://github.com/gioblu/PJON/wiki/ATtiny-interfacing **without** external oscillator.

I can run another examples.

gioblu, 2017-11-21:
Ciao @dontsovcmc what is the result of your connection test? Is that working without external oscillator?

dontsovcmc, 2017-11-22:
Connection 'OK' **without** external oscillator. I choose "Digispark 16mhz - no usb" in Arduino IDE.

gioblu, 2017-11-22:
@dontsovcmc thats a good news :) could you try networkAnalysis ?

dontsovcmc, 2017-11-22:
@gioblu tomorrow check NetworkAnalysis

dontsovcmc, 2017-11-22:
# NetworkAnalysis
Digistump Digispark (Attiny85, uploaded by"16mhz - no usb") without external oscillator, use pin 2
Arduino Micro, use pin 14

# SWBB_MODE 1
## content length 10 bytes

Attiny85:
PJON_PACKET_MAX_LENGTH=20 
PJON_MAX_PACKETS=0

**Test runs 1750 times:**
Fails: 0 - 1 - 2 - 3 
Tests: 1321 - 333 - 82 - 12
%: 75 - 19 - 4.6 - 0.6
sent: 147000, received: 146299, success: 99.52%, failed: 0.47%
```
---------------------
Packet Overhead: 7B - Total: 588B
Maximum Bandwidth: 1428B/s
Data throughput: 840B/s
Packets sent: 84
Mistakes (error found with CRC): 0
Fail (no answer from receiver): 0
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
---------------------
Packet Overhead: 7B - Total: 574B
Maximum Bandwidth: 1394B/s
Data throughput: 820B/s
Packets sent: 82
Mistakes (error found with CRC): 0
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
---------------------
```

## =content length 20 bytes
Attiny85:
PJON_PACKET_MAX_LENGTH=30 
PJON_MAX_PACKETS=0

**Test runs 589 times:**
Fails: 0 - 1 - 2 - 3 
Tests: 478 - 105 - 5 - 1
%: 81 - 17 - 0.8 - 0.1
sent: 34162, received: 34044, success: 99.65%, failed: 0.34%
```
---------------------
Packet Overhead: 7B - Total: 406B
Maximum Bandwidth: 1566B/s
Data throughput: 1160B/s
Packets sent: 58
Mistakes (error found with CRC): 0
Fail (no answer from receiver): 0
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
---------------------
Packet Overhead: 7B - Total: 399B
Maximum Bandwidth: 1539B/s
Data throughput: 1140B/s
Packets sent: 57
Mistakes (error found with CRC): 0
Fail (no answer from receiver): 1
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %
---------------------
```

gioblu, 2017-11-22:
ciao @dontsovcmc this seems meaning all works nearly perfectly, the occasional single failure could be related to the result print itself. Thank you for the support! I will try it too as soon they will be delivered.
That is a great news :)

closed 2017-12-29 by gioblu