gioblu/PJON #27: Better CRC?

TeeTrizZz, 2016-02-28:
Currently, the CRC doesn't take the position of a byte into account. This can lead to problems when bytes are missing (e.g. when a device starts to read the stream at an unknown time and misses the first few bytes). I used my little app to test messages with 5 bytes of content (8 bytes in total). With the current CRC calculation method there were a lot of potential issues (and I mean **A LOT** if one uses more than one device). 

Example: The message 3-8-2-6-4-15-13-9 could be received by a device with the id 6 as 6-4-15-13. The CRC is correct, the device sends a faulty ACK, and the device 3 gets a wrong message because two bytes are sent over the channel at the same time.

The CRC calculation needs to be very simple, I guess. Any ideas?

gioblu, 2016-02-28:
Hi, I don't think what you are describing is a problem of the CRC, it is not failing, it is working correctly, the problem I think is upstream. Also if you bring CRC more convoluted on byte position you can't avoid to get that correct number in the correct position in the pseudorandom bytestream. The responsability to identify if a chunk of the stream is a packet for a certain device I think is of the standard itself, and in this case seems it could lack a packet start symbol. Adding some overhead maybe a start symbol could save some time to listeners that could bounce all bytes are not a start and do something else, and I think reduce much the chance to have a false positive from an arbitrary chunk of pseudorandom bytestream. ie: (lets say 99 is our start symbol)
`99-3-8-2-6-4-15-13-9` so device 6 is safe from false positive
`99-3-8-99-6-4-15-13-9` yes this can happen
`99-3-8-99-6-4-15-13-9` but after 6 should come a number between 4 and MAX_PACKET_LENGTH
and obviously after that should come a series of bytes long as length with the correct CRC value in its place. 

Could you please run your app to see what comes out with a setup with start symbol?

TeeTrizZz, 2016-02-28:
You are right! The CRC shouldn't be responsible for something that shouldn't happen in the first place. I thought about a starting symbol a while ago, but I don't like the idea of using a specific value for this. There will be still a lot of cases where this will go wrong, I guess. How about using the synchronization bit (SyncPad) instead? Something like 1 01 is the start of a message while 1 00 is part of a message (or the other way round). Could this work?

I need to improve my program so the length and the content of the test messages are more flexible. I will probably upload the source then :)

gioblu, 2016-02-28:
I also have thought to syncpad to solve this problem but doing so you are only moving the probblem from byte to bit level, with I think few improvements. I think in this case your app could be really useful to determine how often this could happen with a certain setup. I would try to make some experiments and find the best setup to solve this problem.

gioblu, 2016-02-28:
@TeeTrizZz you added in your simulator also the empty spaces where the channel is not used as a case in the bytestream?

TeeTrizZz, 2016-02-28:
I will do that. I will probably rebuilt the simulator in C# so that timings and things like the syncpad are way easier to simulate. Each device could be a separate thread which reads from a common "wire variable".

gioblu, 2016-02-28:
So my test result is positive with 1.1 I don't have a freeze with a couple of devices in a 24 hours timeframe. Mauro tested 1.1 with 6 nano and 1 mega and all seems fine after 12 hours. Are you sure some of the issues you encountered were not a consequence of using the old code base, some bug in your sketches or some hardware issue?

gioblu, 2016-02-28:
@TeeTrizZz I am doing a simple test to identify if this problem occurs and how often.

I use only two devices, one is always in reception and one is always transmitting 30 bytes packet of random generated data, to a device that is not connected. This leads to message transmission attempts until `MAX_ATTEMPTS` is reached and packet deleted. After this data is generated back and a new packet is created in a continuous loop.

The receiving device will lightup LED 13 when an `ACK `is returned from `receive()` (this means a false positive). I will leave it for 24 hours if I will find the LED on it means that at least once a false positive occured.

TeeTrizZz, 2016-02-28:
Thanks! That sounds good! I'm going do something similar with the six devices as soon as they are back in my office tomorrow. You might be right that there are some other issues in my code. It was pretty strange. Sometimes it worked well with all devices, sometimes only a few devices worked reliably and sometimes nothing worked. I was sending ping-like messages which should be answered by every device in the network. I didn't get a CONNECTION_LOST error but I didn't get an answer by the devices either. It could also be the serial connection or some other bug in my code. I don't know yet. I tested it with three devices on Friday. This worked very well.

Anyway, I am very curious to see where and what the bug is :-)

TeeTrizZz, 2016-02-29:
Unfortunately, we have to move the test to Wednesday. 4 of 6 Arduinos are dead, I have to order new ones. I will come back to you later this week!

gioblu, 2016-02-29:
Glad to confirm you that the LED 13 test described above gave a positive result. LED 13 off = no false positives in a 24 hours timeframe. I will wait for Mauro's 7 devices test result to confirm a third-party problem (see your sketches, hardware wiring, faulty components).

TeeTrizZz, 2016-02-29:
They are from SainSmart. They shouldn't be too bad but who knows. It's good to know that your test results were positive!

gioblu, 2016-02-29:
Buy the blackone and hope :) arduino nano doesnt exist as "original" is not any more part of the Arduino / Genuino products they moved with Micro to 32u4

TeeTrizZz, 2016-02-29:
I guess the main problem was a buffer overflow. I didn't check for that error and I didn't realize that I need to send 17 messages at once. I think I have raised the max value to 15 - which is still not enough of course. That would explain why it worked in some cases and didn't work in other cases. If the channel was free, this was no problem. If the channel was busy, however, then the queue ran full.

I'll check it on Wednesday!

domonetic, 2016-02-29:
@gioblu @TeeTrizZz 
Creo que no es buena idea usar malloc en ningun microcontrolador. ¿porque?
Aqui la explicación http://blog.benoitblanchon.fr/arduino-json-parser/

> What's wrong with malloc() ?
> 
> malloc() make dynamic memory allocation on the heap. It looks for free space and returns a pointer to it, or NULL if not enough continuous space is available.
> 
> free() will make the memory area available for another allocation.
> 
> The problem is that this will create "holes" in the memory and these holes are likely to be too small for the next allocations.
> 
> This phenomenon is called "fragmentation" and it can be solved with virtual address space and defragmentation. This is something that can be done on a computer but not on an Arduino.
> 
> Animation of memory fragmentation
> 
> Moreover, memory allocation on the stack is way faster and easier to predict. If all your allocations are on the stack you know that the result will be the same each time you run the program, you don't have to worry about a malloc() returning a NULL.
> 
> Bottom line: never use malloc() on a embedded system.

domonetic, 2016-02-29:
Perhaps because of this failure , in different scenarios .
In a simple example, malloc() always returns the same location. But in another environment with timers and other objects being created and destroying it, malloc() returns other positions and begins to fail

gioblu, 2016-03-01:
Hi @domonetic, thank you for your support. I totally agree with your point on `malloc()` and `free()`. For this reason I have used vastly the stack memory for all the rest. For the content of a packet to be sent I use `malloc()`, catch its error in case of `NULL` return and free the memory when sent. Because of code's simplicity, there is no chance for the system not to free one of these, and their persistence in the memory is quite short. Considering its variabile length I also see problematic to manage its memory allocation efficiently in the stack, for example not knowing its length at compile time (packet rounting ecc.). On the other side using heap you are also leaving free space for the code! 

Which solution you would use instead?

gioblu, 2016-03-01:
@domonetic What you mean with "begins to fail"? If you consider a failure a `NULL `return, this is covered by the functionality. In the actual master If malloc() returns `NULL ` a 102 `PACKETS_BUFFER_FULL` error is thrown. 
Thank you again.

domonetic, 2016-03-01:
I'm not saying json begins to fail . I say can happen holes remaining , and no scope to deliver the necessary memory and then start to not to send .
As I understand the library should only send and receive packets not keep a list .
The main program should bring the package list.

Please do not think I do not appreciate your work.

gioblu, 2016-03-01:
I didn't know how much was bad seen malloc() and how much problems created in the Arduino environment to people. I agree with you that a static choice could be more reliable and for this reason in 1.2 I will probably as you correctly suggested, switch from dynamic to static memory allocation. 

The only sentence where I am not sure i am catching your point is that the package list should be in the main program. You don't like `packets` array "inside" the PJON object?

Thank you so much for your support and I do not think you don't appreciate :)

gioblu, 2016-03-01:
@domonetic The packet list is there to:
- Save packets are not received correctly and resend them until correct communication occurs
- Save packets to be sent cyclically and manage the sending in the correct time
- Save packets to be sent in future

TeeTrizZz, 2016-03-03:
Everything works fine! I've raised the max value to 25, updated PJON, and checked the hardware. Seems to be very stable now. Next trade show is in London next week. Will come back to you after this event! Thanks for your help!

gioblu, 2016-03-03:
Ok, I will close this issue for now

closed 2016-03-03 by gioblu