gioblu/PJON #32: It's parity, not CRC

jayeye, 2016-03-10:
What you are implementing is a simple parity scheme. Please don't call it a CRC. In any case, there is plenty of spare cycles to implement CRC8 if all you want to use is eight bits.

gioblu, 2016-03-10:
@jayeye You are right, its not Cyclic Redundancy Check, but XOR based error detection.
I will soon fix this issue in documentation and comments. Thank you.

At the actual stage of development this error detection implementation it actually never failed (with a false positive) in 4 years of development, and I never heard by other users of an event like this.
If I have to be sincere in some way I am waiting for a false positive report since years :) to finally implement some more robuts (but also memory and computation thirsty) error detection algorithm. But I think this will happen soon, because PJON is more and more used with long packets, outside of the initially defined "operational envelope".

TeeTrizZz, 2016-03-11:
The communication between my Arduinos is very stable now :-) No false positive so far.

gioblu, 2016-04-23:
CRC8 table-less implementation will be added in 2.0 stable.

closed 2016-04-23 by gioblu