gioblu/PJON #33: Wiring Diagram ESP8266 - Arduino

MichMich, 2016-03-14:
Since PJON is now compatible with the ESP8266 (which is a 3v3 device), it might be helpful to supply a wiring diagram in the readme on how to connect a 3v3 device (ESP8266) to an 5v device (Arduino).

Keep up the good work!

domonetic, 2016-03-14:
simple. you need use a shift level. or zener to drop down.

https://dl.dropboxusercontent.com/u/1641023/ESP8266%20BASIC.png

Please if you need help send me email.

gioblu, 2016-03-14:
Hi Mich, domonetic, I would make a wiki page for ESP.
@domonetic would you help me a little with this part?

gioblu, 2016-03-14:
Ciao @MichMich I had success with these modules:
[image]
[image]

As suggested by user @240974a You just need to put pulldown resistors 1-2 kOhm to signal lines on both sides (3.3v and 5v). So low value resistors because there 10kOhm pullup resistors on the converor board. If you remove pullups from the board, it's possible to use 10 kOhm pulldown resistors.

domonetic, 2016-03-14:
@gioblu yes of course . I help tou

itsjustvenky, 2016-03-15:
Using level shift is always better because the data transmission is faster on level shifter compare to a voltage divider or zenor diode setup.

gioblu, 2016-03-18:
Ciao @domonetic .
I am trying to understand how to give you access to the wiki, so we can work together to the ESP8266 interfacing page. Will write you soon

domonetic, 2016-03-18:
simple. i write a new entry and you accept changes

gioblu, 2016-03-18:
Thank you! Following the actual index standard the title is "ESP8266 interfacing".
I really appreciate your support.

gioblu, 2016-03-26:
CIao! aperepel created a wiki page, I also worked a little on that and this is the result:
https://github.com/gioblu/PJON/wiki/ESP8266-interfacing

Lets me know if you think is enough or you would add something else.

MichMich, 2016-03-26:
Cool! Maybe You could add the zener diode version as welk.

gioblu, 2016-03-26:
Ciao @MichMich I am studiyng the alternatives, and will practically try.
I will then add all the acquired info in the wiki!

Hope you are having good tinkering.

MichMich, 2016-03-26:
I will start my test setup for my PJON curtain system in a few weeks. If I run into nice solutions, I'll add it to the Wiki.

aperepel, 2016-03-29:
Hi @gioblu I've made some corrections to the resistor values (all hail the new shiny logic analyzer for this!) :)

The Uno side requires 1.5K pulldown (2.2K will most probably work as well), otherwise the pin flips high after ACK/NACK. Note that 10K pulldown here results in some noise and, as a result, retransmissions.

The esp8266 is ok with 10K pulldown.

Ran network tests with either board being a sender, all stable now.

High-speed mode **doesn't work** in this setup, only compatibility mode (default). I've updated my previous Fritzing project here: http://fritzing.org/projects/pjon-with-uno-and-esp8266nodemcu

Test results:
```
Absolute com speed: 1867.20B/s
Practical bandwidth: 1556.00B/s
Packets sent: 778.00
Mistakes (error found with CRC) 0.00
Fail (no answer from receiver) 0
Busy (Channel is busy or affected by interference) 0
Accuracy: 100.00 %
```

gioblu, 2016-03-30:
ciao @aperepel, in PJON.h are still not defined timing values for the mode COMPATIBILITY_MODE false for ESP. In any case my suggestion if you want to work on timing tweaking to work on the branch I personally sent you and some other contributors, 2.0 is much more organized and easy to work with timing. I will soon also create the branch as you correctly suggested.

I updated wiki according to your results. Thank you so much for your support.

aperepel, 2016-03-30:
Thanks @gioblu . On the same wiki, the GND to GND connection for esp8266-esp8266 made no difference in my tests. Literally 1 wire was ok. Maybe make a note there, it could depend on how boards are powered.
Now that I've got the right tools, I might look into tuning/adding Fast & Overdrive modes. In the future :)

gioblu, 2016-04-13:
We can consider the wiki page now satisfactory?

closed 2016-04-25 by gioblu