gioblu/PJON #34: Add an option to invert the 0/1 logic so we can use internal pull-up resistors

aperepel, 2016-03-19:
Hi, I was getting frustrated not able to make basics work, but was finally able to dig up that bit in a troubleshooting guide about a 'Busy' signal and, after adding a 10K pull-down resistor, everything worked. I, however, see couple problems with a pull-down resistor requirement:

1. It's an external component. If we are going for simplicity, not the right direction.
1. Most Arduino/Wiring-compatible boards have _only_ pullup resistors (unlike Mega, MBed, etc.). Can we use that instead. The only change required is to invert the logic for reading 1 and 0, if I understand things correctly.

We could e.g. have an overloaded constructor taking an optional pullup flag (defaulting to false).

domonetic, 2016-03-20:
Atmega hace pull-up and pull-down options

domonetic, 2016-03-20:
Atmega have pull-up and pull-down options. No need external resistors 

-----Mensaje original-----

Enviado el: ‎19/‎03/‎2016 20:42
Para: "gioblu/PJON" <PJON@noreply.github.com>
Asunto: [PJON] Add an option to invert the 0/1 logic so we can use internalpull-up resistors (#34)

gioblu, 2016-03-20:
Hi @aperepel. This proposal is interesting on two different sides. What you are saying is correct, and, what you also get adding this, is to have the chance to potentially power the connected devices, as 1-Wire does, using the transmission line (adding the right capacitor and imposing some limits on sending a lot of byte 0 one after the other). 

If you have the will and some time to experiment on a working test version, as you were describing in your message, I will review :) Thank you for your feedback.

@domonetic thank you also for your feedback.
But are you sure of what you are saying? 
see https://www.arduino.cc/en/Tutorial/DigitalPins
Please try not to spam 4 messages with the same context one after the other to mantain the conversation clear and straight to the point :)

gioblu, 2016-03-20:
Here an interesting document on Parasite Powering:
http://www.midondesign.com/Documents/PARAPWR2.PDF

aperepel, 2016-03-20:
@domonetic I bet some confusion crept in. There is Arduino Mega board (2560 chip), which has both pull up and pulldown. Then, the whole Atmel line of chips all starting with AtMega. Uno and Leonardo use e.g. AtMega328P and AtMega32u4, which don't feature pulldowns.

domonetic, 2016-03-20:
You're correct

-----Mensaje original-----

Enviado el: ‎20/‎03/‎2016 12:21

CC: "Maximiliano Duarte" <maximiliano.duarte@gmail.com>
Asunto: Re: [PJON] Add an option to invert the 0/1 logic so we can useinternal pull-up resistors (#34)

gioblu, 2016-03-20:
@domonetic @aperepel didn't know that 2560 includes a pulldown. This is really interesting and maybe should be documented in the wiki how to use it, so the end user will be able to simply set it in his sketch instead of physically adding it.

joaand, 2016-03-20:
The datasheet [http://www.atmel.com/Images/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf](url) does not mention pull downs, only pull ups.

gioblu, 2016-03-20:
Thank you @joaand

itsjustvenky, 2016-03-21:
Even ESP8266 has only internal pullup

gioblu, 2016-03-26:
I have considered a lot, and digged and I obtained as a result only few questions: 

1. 0s are detected as 1s because the channel is "floating" or electricity around the channel is not correctly grounded, to ground this additional unwanted interference pull-down are used, if we invert, we will only experience the problem in the opposite side?
2. And also how we can mantain multimaster capability intact with a base of 1 in a non-interrupt driven system and only 1 wire (already in use) and more than one device pulling up?

closed 2016-04-13 by gioblu