gioblu/PJON #36: Put timing values in a separate file

gioblu, 2016-03-26:
Ciao! I am considering to move timing values in the file `Timing.h` to have a dedicated space for that and to mantain clarity and add practical space for useful comments. Let me know what you think about it.

aperepel, 2016-03-26:
I'm  in support of making things more accessible. This might help. E.g. if one is thinking of writing a custom parser for a logic analyzer capture,  having all possible timings easily accessible helps. Today it's  somewhat buried in main files.

gioblu, 2016-03-28:
@aperepel right, buried is the right word.
I would make this modification after the release (soon) of 1.2 with ESP and Zero compatibility. We are working to test it extensively because its quite a huge update. After all this process I will start with this issue on the next release (1.3).

gioblu, 2016-03-30:
@aperepel I close this issue because has been fixed in 2.0

closed 2016-03-30 by gioblu