gioblu/PJON #38: Implement FAST mode for esp8266

aperepel, 2016-03-30:
Currently there are timing definitions for the `COMPATIBILITY_MODE` only.

sticilface, 2016-04-16:
I'd also like to ask for 160Mhz support.  I always run it at that speed, would that be difficult to add?

gioblu, 2016-04-17:
Ciao @sticilface, if you want to try to implement it, is not so complicated:

- Look in Timing.h, add the 160Mhz condition following the file format
- Add the new timing setup.

To find the correct timing you need only to be sure to have a functional setup with the STANDARD mode between an Arduino and an ESP. If you have a way to measure the bit duration of your ESP in STANDARD mode, you will only have to find the timing setup to obtain the same result when running at 160Mhz. Start from the 80Mhz setup. Probably it will have to be slightly tweaked.