gioblu/PJON #39: MSB/LSB switch?

aperepel, 2016-03-30:
Hi, I spent some time debugging the physical comms on the wire, and saw that current PJON version uses LSB-first mode. This is quite inconvenient when looking at signals and trying to make sense of them. Was it done for some performance reasons? If not, does it make sense to ad the MSB/LSB option and leave LSB the default?

gioblu, 2016-03-30:
Ciao @aperepel, you are right. After long tests I found the send_byte function algorithm as the fastest to be executed. If I am not mistaken also Serial and SoftwareSerial are using the same approach (LSB-first). I know I had the same issues when looking for the first time to it with an oscilloscope, but after a while i got used to it (initial padding bits helps a lot human readability). I think the use of LSB first approach  in any case is an information should be noted in readme. I will add it soon

gioblu, 2016-04-01:
Infomation included in the home wiki page. Thanks for reporting.

closed 2016-04-01 by gioblu