gioblu/PJON #40: Phantom device

thomsna, 2016-03-31:
Two Arduino pro mini ATmega328.
A master set at address 45 and a slave set at 44. 
Bus on pin 12.

While sending to device 124 even though it didn't exist on the network, I get ACK values back.

Can someone reproduce this?

Code for master:
```C
#include <PJON.h>

PJON network(12, 45);

char content[] = "ajtyenflhiehfnvk";

void setup() {
	Serial.begin(115200);
	delay(250);
	uint8_t test = 0;
	long scan_time;

	for (uint8_t i=0; i<256; i++) {
		scan_time = millis();

		while (millis() - scan_time < 20) {
		int response = network.send_string(i, content, 16);
			if (response == ACK) {
				test++;
			}
	    }
	if (test > 0) Serial.println(i);		
	test = 0;
	}
}

void loop() {};
```

Code for slave:
```C
#include <PJON.h>

PJON network(12, 44);

void setup() {
  Serial.begin(115200);
};

void loop() {
  network.receive(); 
};
```

aperepel, 2016-03-31:
@thomsna if I read the code correctly, address 124 is the broadcast. In your teat, try counting how many acks you get and from what addresses to confirm.

thomsna, 2016-03-31:
@aperepel Thanks for clearing that up. 124 is indeed the broadcast address. 
The code above will check for the ACK from a device on all addresses. An ACK is received on address 124, even though no device is set to that address.

aperepel, 2016-03-31:
Great, this is not an issue, it looks. I would maybe prefer to have the broadcast on a more 'classic' address like 0 or 255, than in a middle of the range. @gioblu any comment on that? 124 doesn't  even look cool in binary :)

thomsna, 2016-03-31:
+1 for setting broadcast address to 0, like I2C :)

gioblu, 2016-03-31:
Hi @aperepel @thomsna to be consistent and not subjected to versioning changes (as the one you are proposing) bugs, always use constants if you can: `for(uint8_t i = 0; i < 256; i++) if(i == BROADCAST) continue;` so if `BROADCAST` value will change during development this will not affect your sketches.

Could be more clean I agree, we should run extended tests, but if all is ok, I don't see limits in changing BROADCAST to 0. Thank you again for your support and passion in this project :+1:

gioblu, 2016-04-07:
Ciao @aperepel, @thomsna.
With the release of the 2.0 version this issue has been fixed and BROADCAST id is now 0, as correctly proposed :+1: . Thank you again for your support and passion in this project.

closed 2016-04-07 by gioblu