gioblu/PJON #41: PJON Protocol Analyzer

aperepel, 2016-03-31:
Hi guys,

I've been busy hacking recently (well, scratching my own itch), and brought a custom protocol analyzer for Saleae Logic to a usable state: https://github.com/aperepel/saleae-pjon-protocol-analyzer

It doesn't yet understand multi-byte payloads (will come next, so for now don't send more than a 1 byte payload :) ). It can do a lot already, and I figured I'd put it in front of people sooner rather than later:

https://github.com/aperepel/saleae-pjon-protocol-analyzer/releases

Happy hacking!

gioblu, 2016-04-01:
This is really fantastic. :+1:  
The only thing I can say, I will buy tomorrow the hardware and I will be soon on the way testing it.
Thank you so much creating a so amazing debugging and analysis tool for PJON.

gioblu, 2016-04-01:
Lets add it in the wiki!!!

aperepel, 2016-04-05:
Hey guys, I've just made public a massive release. Theread are now builds for OSX, Linux and Win64, which should reach most people. Give it a try with e.g. examples included in the `assets` folder of the repo. https://github.com/aperepel/saleae-pjon-protocol-analyzer/releases/tag/0.6.0

gioblu, 2016-04-13:
@aperepel Compliments. Reading your progress.

gioblu, 2016-04-13:
@aperepel Where you think would be more appropriate to showcase this tool for troubleshooting or analysis?

aperepel, 2016-04-13:
Analysis for troubleshooting :) It's a great tool to visualize the actual protocol, but also helps find timing issues once things are connected. I'm leaning more towards analysis, as it will be not the only tool for troubleshooting, but an important one in a toolset.

gioblu, 2016-04-13:
Fantastic, compliments again. Maybe my english overflowed :) but I meant where we could embed a link to your work inside the PJON's repo documentation / wiki

aperepel, 2016-04-13:
Sure, I see. I would personally link to it somewhere from an introductory page where it describes the protocol in details - e.g. it helped me immensely once I saw what was going on on the wire. I found that troubleshooting page is very useful, but has tons of information, the link could be lost. Or maybe link from both :)

gioblu, 2016-04-13:
> If you have a Saleae Logic Analyzer you will be amazed by saleae-pjon-protocol-analyzer crafted by the user aperepel, it offers a complete analysis suite aware of the Standard functional procedures and symbols.

https://github.com/gioblu/PJON/wiki/Troubleshooting

gioblu, 2016-09-08:
Ciao @aperepel how its going?
Have you considered to update the analyzer to be compliant with the 0.3 specification?

gioblu, 2017-09-04:
Ciao @aperepel how its going? Have you abandoned the analyzer development? If so I will close here and remove it from the compliant tools!

gioblu, 2017-09-07:
closing for now, feel free to open back if needed! :)

closed 2017-09-07 by gioblu