gioblu/PJON #42: ATtiny841 and ATtiny1634 Support

gharris012, 2016-03-31:
I have been able to make this library work (well, the Blink Transmit/Receive examples) on the ATtiny841 (core from https://github.com/SpenceKonde/arduino-tiny-841) by adding __AVR_ATtiny841__ to the list of Tiny chips.
DigitalWriteFast.h needs some updates as well, but they are over my head (I just reverted back to DigitalWrite in setup to get Blink to work).

It looks like adding 1634 support should be pretty similar. I don't have one of those soldered up yet to test on but it did compile.

I have both chips and can do whatever further testing is needed.

On a happy note, this library is pretty exciting! I have been trying to get OneWire to work between an ESP-03 and a Tiny841 for a while now, and this worked the first try. Good stuff!

Thanks!

gioblu, 2016-04-01:
Ciao @gharris012. Add compatibility to this chips would be amazing!
And I am happy you got a fast positive test result.

Please if you can describe your test setup and what you did to get it working (you changed or defined new timing for 841?). If you would propose a code base addition / modification wait some days, we are releasing 2.0 and if you pull request after release, merging will be less painful :)

Thank you so much for positive feedback and support :+1:

gioblu, 2016-04-07:
ciao @gharris012, 2.0 has been released now. Finally there is a more organized way to port new MCUs / architectures. take a look to the new `Timing.h` file: https://github.com/gioblu/PJON/blob/master/Timing.h

Now you can consider to make a pull request with your results on top of the new release.
Happy tinkering :+1: