gioblu/PJON #44: license

hellerbarde, 2016-04-02:
Is the Stipulation about advertising materials compliant with common Open Source license requirements? 
Or is this not meant to be strictly speaking [Open Source software](https://opensource.org/osd-annotated)?

The section I could see being violated is "1. Free Redistribution" because requiring the ad-note could be interpreted as payment of sort.

I don't feel comfortable to say that this is certainly a problem, and I would be glad if a person more competent in these matters could weigh in here.

In any case, your protocol is fantastic :+1: and I'm glad someone has liberated the space of one-wire-ish protocols. I also felt uncomfortable using the dallas one-wire stuff...

Also mad props for the nice and clear writeup of the protocol in the wiki! Well done, sir! :tada:

gioblu, 2016-04-02:
@hellerbarde you are right, this issue has been pointed out. And I agree and hope with you, that someone, with a strong experience with this will give his/her contribution to this project.

This issue is considered to be fixed in 2.0 release and one considered option is to select Creative Commons Attribution.

aperepel, 2016-04-02:
To be fair, current license is very similar to a CPAL license, which is a standard approved OSS license. I agree I would prefer to converge on an easily recognizable license, rather than a customization of one.

gioblu, 2016-04-03:
I am studiyng this field, I perceive this as maybe the most compatible:
https://creativecommons.org/licenses/by-sa/4.0/legalcode

gioblu, 2016-04-03:
Ciao @aperepel the actual license is BSD-3 clause, but I don't remember from which source I took this and seems not to be the Standard BSD-3, thanks for reporting. Will be changed in 2.0 (near to release) to the standard BSD-3. After study, I don't see big limitations for the user or companies to adopt PJON in their works / products.

gioblu, 2016-04-08:
Ciao @aperepel @hellerbarde , thanks for reporting License Clause has been updated in 2.0 version as proposed to BSD-3 Standard License Clause.

closed 2016-04-13 by gioblu

gioblu, 2016-04-13:
Switched to Apache 2.0.