gioblu/PJON #45: Can I mess with MAX_PACKETS and PACKET_MAX_LENGTH?

thomsna, 2016-04-03:
In order to free some memory, I see that reducing MAX_PACKETS and PACKET_MAX_LENGTH helps a lot.
Do they have an influence on the length of the packets I send? I'm for example never sending packets larger than two bytes.
What should I consider if I want to reduce those values?

gioblu, 2016-04-03:
I am happy to see you understanding so clearly this part.
Yes you can freely tweak this values to your needs. Remember that a string longer than MAX_PACKETS can't be received or sent by the device.

thomsna, 2016-04-03:
Don't you mean PACKET_MAX_LENGTH? 
If I understand this correctly: if packets are never longer than two bytes, I could set this value to 2.

gioblu, 2016-04-03:
Yes I am sorry, `PACKET_MAX_LENGTH`.
Absolutely. This could obviously also co-exist with other devices transmitting bigger packets.

gioblu, 2016-04-04:
Also `MAX_PACKETS` value tweaking affects memory.
Less packets the system has to handle less memory is consumed.

Consider how many packets you need, in the worst case, to handle in the same time and set it as `MAX_PACKETS`. If you don't have other questions I would close this issue.

Thank you again for your interest.

thomsna, 2016-04-04:
No other questions on this subject :)
Thanks for the clarifications.

closed 2016-04-04 by thomsna