gioblu/PJON #46: easy function to send Integers

littleprojects, 2016-04-05:
Hi, 
is there a easy function to send and receive a integer value (8,16 or 32 bit). I think, this is the most payload. And its easier to handle than a string of chars. 
And in combination withe the sender_id as a content_id it would be more awesome. So i can send one command to all devices.

gioblu, 2016-04-05:
Ciao @littleprojects. One of the IDs is dedicated to `BROADCAST` to all devices. Every device will receive the message but will not respond with ACK. This gives you a slight performance gain but you loose the certainty of packet delivery. Maybe some of them are busy executing something and will not receive the message and you are not aware of it, so it's useful for some key situations like general assert to all devices repeated every 5 seconds where you don't need reception certainty.

I wil take a second to reason on your proposal of content_id to have a proper answer to you.

gioblu, 2016-04-05:
If I understood correctly you would be able to pass to the `send()` function an integer, without having to pass also its length? I think this can be done to ease the transmission side sketch. But, I am not sure on the receiver part would be possible to rebuild the int in "background" without adding a considerable amount of complexity. But, obviously, if you have a better idea, share it :)

littleprojects, 2016-04-05:
maybe like this:
```
send_int8(id, int8_t);
send_int16(id, int16_t);

reseive_int8(data[0]);
reseive_int16(data[0]&data[1]>>8);
```

gioblu, 2016-04-06:
Maybe something like: 
```cpp
int send_int16(uint8_t id, int data) {
  return this->send(id, data, 2);
}

// so

network.send_int(44, 10000);
```

On the receiver side I would propose a set of functions able to "parse" content in the receiver function. But I think they should not be part of the PJON class, but more probably a dedicated class in the includes directory? I don't think is something should be included in the PJON class.

dterweij, 2016-10-19:
Would that be my problem?
How do i send this:

`
String celsius;
celsius = (float)raw / 16.0;
// example:   the string holds:   celsius = "23.36"

bus.send(44, "Hello" , 5);
`

How to send celsius without  the compile errors (mixing/converting int/string things)...

gioblu, 2016-10-20:
Ciao @dterweij would be nice to add support to string type.
We are now in the process of release a new version. Will soon dedicate some time to this, if you have a fix proposal feel free to share. Thank you for pointing out.

dterweij, 2016-10-20:
Right now I solved it this way:
```
celsius = (float)raw / 16.0;
 float T;
 char TStr[9];
 T = celsius;
 dtostrf(T,5, 2, &TStr[0]);
 bus.send(44, TStr, 9);  
```

But as a add more temperature sensors later I have to send a ID like:

A 12.45 B 23.34 C -2.53

Now that my sender is working, I am working now at the receiver Arduino with Ethernet Shield. It directly stores temperatures into a MySQL database.

I needed to use 2 Arduino's because the receiver did had a to big sketch. (mysql, ethernet, ntp  libs).
Soon I connect a serial driven sensor board to the arduino to read out 4 sensors. Because i want to free up my laptop because its the only computer left with a serial com port. 

I am not skilled to help.  I do just read a lot and use copy paste code for a desired result. If not working I go to the next try :)

I'm using version PJON 5.0.0 because above this has compile errors. 5.0.0 shows errors too but at least it works :)

gioblu, 2016-10-20:
Ciao @dterweij on which architecture / platform you are using PJON?
Which errors you get? Could you document them in issues to lem me fix them? :)

gioblu, 2018-01-13:
I close here for now, feel free to reopen if required

closed 2018-01-13 by gioblu