gioblu/PJON #47: Bootloader to code Arduino through PJON

gioblu, 2016-04-07:
Ciao! I am a while thinking about this. Would be amazing to ble able to code any device on a PJON bus, through PJON and the bus itself using a dedicated bootloader.

sticilface, 2016-04-30:
Would I be right in thinking that if this was possible on an arduino you could upload sketches to it by pjon. This would be amazing, as I planning to deploy a load of custom arduinos, on custom pcbs (not designed yet) but reprogramming is an issue!!

gioblu, 2016-04-30:
Yes correct, would really be an amazing functionality, I am digging. .

ictabc, 2016-05-24:
Have a look at [Tine Safe Boot](http://jtxp.org/tech/tinysafeboot_en.htm). Maybe that will give you some ideas.

gioblu, 2016-06-01:
@ictabc this is really cool, thanks!

dtony, 2017-01-25:
This would be a really good feature! I thing https://github.com/mharizanov/arduino-netboot is a really good starting point. The code does not look really complicated and it also looks like most of the code is handling the Ethernet module.

I would like to have a look at it but I am not sure to have time and skills to do it.

JohnKiller, 2017-04-23:
To make a bootloader you have 2 main problems. First, tell the compiler that your program doesn't start at address 0 but it is located in the last pages of the ATMega program memory. To do that, add `-Wl,--section-start=.text=0x7000` to the compiler command line. I don't think you can do this with standard arduino IDE, so I've tried using [bare arduino project](https://github.com/ladislas/Bare-Arduino-Project). It compiles OK, but here we go to the second problem: space. Bootloaders are small, because the space reserved to them is controller by 2 fuse bits, and the max you can get is 4096 bytes.

The library alone compiled in this way (and with max 1 packet, max 64 bytes) is 5154 bytes (with `-Os` flag). Most of this space is also because the Arduino functions are very huge (`digitalWrite` is like ~300 bytes).

So I've removed the Arduino header files, replaced all the required functions with `NOOP`s and the size is now down to 3404 bytes. But I have to implement those in bare C++ with only avr functions, and I don't know how to do that.

If anyone wants to help, I will upload everything to GitHub.

(PS. also I was using `SoftwareBitBang` because it's the most useful feature of PJON, with only 1 wire required and no extra hardware, but i've noticed this is also the biggest, along with `AnalogSampling`. Probably this whole thing can be done a lot better with `ThroughSerial` and a hardware serial)

JohnKiller, 2017-04-23:
Probably this depends on #89

gioblu, 2017-04-23:
Ciao @JohnKiller avoid the use of `send`, and use only `send_packet`, so not making use of the packet scheduler reducing memory footprint (so not needing to call `update`). For sure another thing could be to reduce the basic arduino functions footprint. `SoftwareBitBang` has pros and cons compared to `ThroughSerial`, depends what is your requirement and what you need to accomplish. In any case I see this issue as important for many and extremely interesting to tackle, so if you need help feel free to ask, I would be happy to support you.

JohnKiller, 2017-04-30:
Ciao @gioblu I've created a [repository here](https://github.com/JohnKiller/PJON-bootloader).

I've successfully compiled the program using your suggestions and I'm about ~3000 bytes in size.

There is a lot of work still to do, I'm working on it.

gioblu, 2017-04-30:
ciao @JohnKiller this is an amazing work! Compliments! Feel free add me on gitter or to visit PJON gitter chat if you want to have a sync discussion :)! Would be nice to have a chat.

Is PJON-bootloader able to program one certain device in a bus with many others connected?
Have you tested it?

Compliments again.

JohnKiller, 2017-04-30:
Thanks @gioblu, once finished it should handle multiple devices with no issues, since most of the work is done by your library.

What I did initially was to create a bootloader from scratch using USB serial, and that is working so now I'm porting that over to PJON