gioblu/PJON #49: Refactoring digitalFastWrite.h include to be a sibling

aperepel, 2016-04-12:
Hi, can we mover the subj to be on the same level as all other header files? E.g. it makes things work out of the box when not using an arduino ide, otherwise have to go in and modify the paths manually.

gioblu, 2016-04-12:
You would remove includes directory?

aperepel, 2016-04-12:
Yes, it always caused issues when building from command line. E.g. with PlatformIO, the library becomes available on a system path via <> style includes. Using directory paths in "" doesn't work well with this structure.

gioblu, 2016-04-13:
Ciao @aperepel I am a little captured by some issues, could you please propose a fix to this?
Or in any case I will come back to this point as soon as possible.
Thank you again.

gioblu, 2016-04-23:
To be fixed in 2.0 stable

closed 2016-04-23 by gioblu

gioblu, 2016-04-23:
@aperepel fixed.
Now should be more easy to be used on platformio.