gioblu/PJON #51: Explanation about connections

domonetic, 2016-04-13:
I understand that just connect 1 Data Cable works, but is there you got previous care, pullup resistors, diodes protection?
No it is not clear how protect in a noisy environment or outdoor use.
Can add this to wiki?

gioblu, 2016-04-13:
Ciao @domonetic. Hope all is going good.
Abolutely, you are right, maybe I should create a Wiki section dedicated to interference reduction.

domonetic, 2016-04-13:
Yes please. I try to test this lib with RF 100w more than 25 meters away. 
Do you test with Arduino pro micro?

gioblu, 2016-04-13:
Ciao @domonetic, 100w?? You are transmitting to mars :) or you mean mw ?
In any case I would be happy to help you, please if you can describe technically your setup and I will try to support you to get a decent communication channel.

- PJON (not PJON_ASK) is not tested on radio and probably is not effective because is a standard single sample in the middle of the bit, PJON_ASK uses a sort of "oversampling" I would call digital-pseudo-oversampling. 
- PJON is not tested with Arduino PRO micro (3.3v logic level could be a issue)!!! What is making me dubious is that maybe, you are using the logic pin of micro to feed radio tx and maybe the 3.3v produced are not enough to generate the radio waves intensity you need to cover the gap between receiver and transmitter.

domonetic, 2016-04-13:
I do not transmitting to Mars, but yes, to the rest of the world, I am amateur radio. The idea is digitalize a rotor antenna to find the position and display it in my rack. The transmit antenna (interference) is about 50cm cable distance, the control logic will be shielded by the rotor casing. Also as another serious interference that the power line is 24VAC.
I use arduino PROMICRO be 5v.

gioblu, 2016-04-13:
Fantastic, compliments :+1: !!!
Happy to know we have a radio amateur in the PJON community (I am attending the courses to get the call-sign here in Italy!). Do you have some sort of result with communication or you get absolute absence?  Are you using PJON or PJON_ASK implementation?

Possible fixes:
- Different power source
- Diode before power feeding to Arduino
- Pulldown resistor on the transmitting and receiving pin
- Different antenna design
- Tweak values of timing in PJON_ASK.h

gioblu, 2016-04-13:
@domonetic try also to use a Nano/Duemilanove/Uno to clarify if the problem is the board itself or the rest.

domonetic, 2016-04-13:
Great, I am LU4HNE when you have it we connect. Or if you do operational practice let me know.
I do not even armed myself because I have two scenarios to test. Before i need correct a minimal problem with movement of pot.

1. Potentiometer reading the crown
2. magnetometer or external Compass

And send readings to the base station and display

Also, you can send that want to position degrees, and make manual movements

gioblu, 2016-04-13:
Seems really an interesting project. When you will be working on it feel free to ask support here.
And if you want to share it done, soon we will release PJON's website where we will showcase some projects by the users. Happy tinkering.

gioblu, 2016-04-13:
@domonetic soon follows addon on the wiki as you requested.

gioblu, 2016-04-13:
Hi @domonetic , I am still working on it, but you can already take a look: https://github.com/gioblu/PJON/wiki/Deal-with-interference

domonetic, 2016-04-13:
Magnific!!. 
In buses rs485 are used TVS or diodes in opposition
This suppresses transient voltages, used olso in ADC systems

        ^ +vcc
         |    
        [=] DIODE
         |
PIN -------------- BUS
         |    
        [=] DIODE
         |
        --- GND

You think you can work here?

gioblu, 2016-04-13:
Never tested this approach. Can you please explain more clearly? I would be sure to have understood the circuit diagram!

domonetic, 2016-04-13:
[image]
[image]

domonetic, 2016-04-13:
first. This method is used on inputs of home alarms
Second. More eficient protection used in rs485 buses.

gioblu, 2016-04-14:
Ciao @domonetic can you check in the new page wiki that, what you suggested is correctly explained?

domonetic, 2016-04-14:
Good!! im preparing my RF test. when it done i upload a video.

gioblu, 2016-04-14:
@domonetic Fantastic. Feel free to open an issue here to showcase your work. We will manage to add it in the application examples made by the contributors and users like you. Thank you again for your support and passion! (I will for now close this issue because seems we reached a satisfactory result)

domonetic, 2016-04-15:
hear a prototype
https://youtu.be/UtZoIAoqQns

Tomorrow the real test.

gioblu, 2016-04-17:
@domonetic I saw the video, compliments!!!! I am really curious to see the final result.

olehs, 2016-04-17:
@domonetic, the first pircture with protective diodes shows these diodes are already a part of the AVR (which is also described in Atmel's datasheet part about I/O pins internal protection).
As far as I understand, you suggest adding extra diodes for protection, right?

I also heard about PTC and Zener diode protection of IO pins, but don't really know if this approach is suitable for PJON protocol.

domonetic, 2016-04-18:
The PTC do not know if they work, the zener often cut off or slow pulses. It
is best to use tvs or high speed diodes. I think a couple of diodes helps
maintain healthy inputs and is not much to add them.

I simple tests with 15 meters of cable. but still it raining in my city and I
can not even try with rf

gioblu, 2016-04-25:
Ciao @domonetic how its going with the rotor antenna experiment?
2.0 stable is finally out, give it a look! :)

gioblu, 2016-04-25:
Ah and if you can tell me if you consider the wiki page satisfactory and so this issue solved.

domonetic, 2016-04-26:
So far so good, I'm testing the magnetometer gives me a headache, so I open in
case of finding any error.
But with 15 meters of cable does not give any communication problems

gioblu, 2016-04-26:
Good send us a video! I am so curious to see your application! 
Happy tinkering.

closed 2016-04-29 by gioblu