gioblu/PJON #52: Reception strategies

gioblu, 2016-04-13:
Hi there. Is long time I am considering to give the chance to the user to develop its own strategy for data reception, and so enable to switch the low level functions needed to physically receive and transmit data. This would give the chance to the user to be able to port different communication media, Infrared / visible light, radio waves (see PJON_ASK) and obviously one or more conductive elements.

I personally developed 3 strategies in my experimentations:
1. Digital bit centered single sample (very effective on wires / conductive elements)
2. Analog read oversampling (quite effective used with infrared light)
3. Digital pseudo-oversampling (quite effective on noisy wires and radio)

Would be amazing to have the chance to select on instantiation which strategy is more effective with the communication medium used. 

Obviously:
- PJON_ASK would be integrated in the PJON repository as a Strategy
- Timing.h would be definetly more complicated (3 timing setups per strategy)
- Maybe will be difficult to constrain data rate without excluding some use-case.

Makuna, 2016-04-30:
I have some technical advice on how you can achieve this goal.  I will try to contact you out of band for discussion.

gioblu, 2016-04-30:
Ciao @Makuna it will be a pleasure to hear from you! gioscarab@gmail.com

gioblu, 2016-05-04:
Thanks to @Makuna suggestions I am developing the new framework for physical layer strategies based on templates and quite a new codebase, so the user will be able to easily develop its own strategy or use the proposed ones, also instancing multiple instances of PJON with different communication strategies / media

agsb, 2016-05-04:
great news!

agsb, 2016-05-04:
Consider add me to group;

gioblu, 2016-05-04:
Ciao @agsb! Soon will come the release, what you mean about the group?

agsb, 2016-05-05:
hello,

If you need any extra beta tests and followups, maybe some code too, i
would join.

best regards.

gioblu, 2016-05-05:
for sure, I am finishing to document and create correct examples to the version I am locally developing. Soon will come the new release. I need still minor checks.

Sincerely thank you for you support to the project.

gioblu, 2016-05-06:
Ciao @agsb you can find it now unreleased in master, if you can make some tests would be highly appreciated! :+1:

closed 2016-05-06 by gioblu