gioblu/PJON #55: Arduino -> ESP8266 communication.

sticilface, 2016-04-18:
I've been struggling with this for a whole day, and I'm not sure what to do. 

Logic is working. See osillyscope, correct signals, correct voltages.  I've got 2 arduino nano's communicating.  One using BROADCAST, the other receiving. Using STANDARD mode. ESP8266 is 80Mhz.  If I write a sketch that attaches an interrupt it detects the pulses.. and seems like the correct timing... but I have no debug info in PJON to work out why it is not detecting the pulses. 

[image]
[image]
[image]

Any help appreciated.  It looks like the initial low pulse might be a bit long from the arudino... 400us.  but I'm really not sure.

sticilface, 2016-04-18:
So adding a print to the receive byte function gives me this
```
{et{"n","lu:1t{"t""p":"vue1} "s":pi5,al":}
```
and what this should say is this
```
{"set": {"pin":1, "value":100}} 
```

so it is receiving and kind of there... just not quite..

sticilface, 2016-04-18:
OK.. got it working
```
       #define BIT_WIDTH  46
       #define BIT_SPACER 112
       #define ACCEPTANCE 35
       #define READ_DELAY 4
```

gioblu, 2016-04-19:
CIao @sticilface 
So you fixed changing the Timing value.
Which ESP you are using? Which clock and toolchain?
Are the standard examples working with the standard Timing?

sticilface, 2016-04-19:
Need to check. Basically this is receive only and my sketch. Need to test sending. There may be an issue with functions being cached to flash on arduino esp. They are by default. And this causes delay at the start. Icache_ram_attr needs to be added. I'll see though.

gioblu, 2016-04-19:
Ciao @sticilface you are right. I should probably have considered this in my answer to you above.
Please, to be sure all is alright, try the test sketches to be sure to have a stable communication in the standard scenario in the first place.

sticilface, 2016-04-19:
so I've just spent the entire evening battling!!!  I could get the receive to work on the arduino, but just not the send.   major debugging.. took me all through the code... checking the different methods you have for pin writing... which were all fine... in the end it was because the pin was busy... oscilloscope out again... and indeed the ESP would write to the pin when it was pulled down.. but not when connected to the level converter...  to cut a very long story short... I needed to add a 10K pull down to the 5V side, and 1K on the ESP side.  Seems to be working fine now!  I'll see what speeds I can increase to on both sides!  The good news is that I got the interrupt version to work fine, even without disabling them, there is no need to ICACHE_RAM_ATTR seem to work fine without.  I'm going to see how much stress I can put the PJON under whilst running wifi as that is where we might see some issues!  
Thanks for the help

working send and receive timings
```
      #define BIT_WIDTH  46
      #define BIT_SPACER 110
      #define ACCEPTANCE 35
      #define READ_DELAY 4
```

sticilface, 2016-04-19:
I've managed to get the OVERDRIVE mode working between the ESP8266 and an arduino nano
Absolute com speed: 3216.00 B/s |Practical bandwidth: 2680.00 B/s |Packets sent: 1340.00 |Mistakes 2.00 |Accuracy: 99.85 %

but this is only with the ESP8266 in receive.  
If i run the speed test on the ESP8266 I get wdt reset...

gioblu, 2016-04-19:
Hi @sticilface. Yes, the pull-down resistors necessity is a known issue described in the ESP wiki page.
I considered you already passed there I am sorry.

In any case, good result :) 

`OVERDRIVE` mode is there to let the users run the architecture at its maximum speed regardless compatibility, so max speed ESP to ESP. In my opinion if you want to contribute looking for still not implemented timing setups, would be amazing to find ESP to Arduino in `FAST` mode that still is not implemented or to run in `OVERDRIVE` ESP to reach its limits (and so implement `OVERDRIVE` mode on ESP)

gioblu, 2016-04-19:
@sticilface Can you please tell me if you have a correct communication also with the Timing setup is actually in the master version? Why you changed from 44 to 46?

sticilface, 2016-04-19:
I did read the wiki, but i'm just using raw ESP8266, so no extra pull down, as you suggested then requires the 2K not the 10K.  nonetheless I have it working now.. 
ESP8266 to Arduino, and back 
```
Absolute com speed: 1802.40 B/s |Practical bandwidth: 1502.00 B/s |Packets sent: 751.00 |Mistakes 0.00 |Accuracy: 100.00 %
```

so correct me if I'm wrong.  FAST is still platform independent whereas overdrive requires the same platform.  I just thought the ESP should be able to do the fastest, but i see you point.  reserve OVERDRIVE for ESP to ESP coms.  I'll see what I can do. 

regarding changing 44 to 46.  It the only thing that worked.  even 45 doesn't work!  46.. works like a charm.  no idea why.. maybe it is this caching... i know @makuna had issues with it, and had to declare is functions as C 'extern' in neopixelbus... I can ask him if it is still an issue?

sticilface, 2016-04-19:
I've also modified the speediest, to be async, as a 10second while causes wdt.  Using interrupts also seems to work pretty well on the ESP.

Makuna, 2016-04-19:
With the latest Esp8266/Arduino, the ram attribute works even on C++ files.
Something to note, all software pin IO on Esp8266 runs at 8mhz update, you can see this if you toggle a pin in a loop and look at it on a logic analyzer (or oscilloscope that captures).  If you try to go faster, you may loose pulses.

gioblu, 2016-04-20:
Ciao @sticilface now I understand, I thought your setup was NodeMCU. I never tested a raw ESP.
So you proved a raw ESP to be compatible. :+1:  

If you are interested to make a pull request with your changes I will ask you some questions:
- Do you have a NodeMCU, to perform a comparison test and to test mutual compatibilty? 
- Do you think is possible in `Timing.h` to conditionally define timing for both NodeMCU and raw ESP with the presence (or absence of) predefined constants?
- I understand your point on 10 seconds wdt, the test time can for sure lowered to 1 second.

Ciao @Makuna, thank you for reporting this useful new feature. Speaking about the pin update, is more than necessary also for `FAST `mode, will be interesting to shave microseconds and see where is the limit in `OVERDRIVE` between two ESP.

Thank you again.

sticilface, 2016-04-25:
PS.  I've just got PJON receive working on the TEENSY 3.1 (not checked SEND yet.) 

Makuna has mentioned that on AVR, the timer for micros() can have a variable resolution of 2,4 or even 8 micros.  That might be why there is so much change with the micros between boards.  worth a mention, i thought.

gioblu, 2016-04-25:
Fantastic!! Would be amazing to personally test.
Thank you for your support.

Check 2.0 release :+1:

gioblu, 2016-04-25:
@sticilface interesting i lost that about resolution, it's an intelligent consideration.

Makuna, 2016-04-25:
http://www.arduino.cc/en/Reference/Micros
see the end of the first paragraph ;-)

gioblu, 2016-04-30:
Ciao @sticilface, would you share the modifications you made to port TEENSY to add it to the master version?

gioblu, 2016-05-06:
Thank you for sharing :+1:

closed 2016-05-06 by gioblu