gioblu/PJON #56: Auto addressing sometimes finds duplicate addresses

thomsna, 2016-04-19:
The ARP implementation sometimes finds duplicate addresses.
Tested with 5 slave Arduinos (Pro Mini) reporting their address to a master.

I rolled my own ARP for a project – the way I implement it is having all slaves send their randomly generated address to a master, along with a UID (randomly generated 32-bit integer). The master just looks for duplicate addresses if any, and send a new address to the duplicate devices along with the UID. The slave with the corresponding UID will be set to the new address. This works because the slaves have to talk to a master device which handles the addresses.

Having a robust ARP independent of a master device with PJON is perhaps non trivial to do.

gioblu, 2016-04-20:
Ciao @thomsna happy to hear from you here.
So, the issue you are reporting has been detected with extended tests, and this can happen if you startup devices in the same time and they generate a very similar initial delay duration. 

I totally agree with you that with a master slave scenario, ARP is much easier and less critical to achieve.

I am locally testing various approaches. The one that seems more effective is to schedule after id acquisition, a repeated sending to the acquired id, in case of an answer, collision is detected and id acquisition restarted. 

I am not using random because of the really poor randomness of numbers generated by an Arduino.
trying without randomSeed, often the same numbers in the same order, also after a restart. So you can imagine what this could lead to in a busy network. 

With the use of randomSeed the situation is good exactly after boot, but if aiming to an unconnected analog pin this after an itnitial floating state will tend to seat on a static voltage and this leads to poor randomness again.

thomsna, 2016-04-20:
I tried to generate a better seed for randomSeed(), as I need unique 32-bit numbers for my project.

On the Arduino Pro Mini there are 8 ADCs – reading from them all, and giving those numbers to a hash function seems to generate some pretty good data. I was able to generate [~20000 unique numbers](http://pastebin.com/raw/Qc56Dnjc) with no duplicates, until I stopped the readings. 

This function takes consistently 7ms to run, by taking 32 readings on any random analog pin and generating the hash. Taking 16 readings reduces the runtime to 3.5ms and gives similar results it seems.

```C
uint32_t getSeed() {
	uint32_t seed;
	uint8_t i;

	for (seed = 0, i = 0; i < 32; i++) {
		seed += analogRead(random(0,8));
		seed += (seed << 10);
		seed ^= (seed >> 6);
	}
	seed += (seed << 3);
	seed ^= (seed >> 11);
	seed += (seed << 15);

	return seed;
}
```

The hash function used is this one: https://en.wikipedia.org/wiki/Jenkins_hash_function#one-at-a-time

I don't know if it would help with the random delay problem?

gioblu, 2016-04-20:
Yes if it has a good performance could for sure help the random delay problem, and also open up the possibility to switch to random id search strategy instead of the actual implementation, that is for sure slower and less efficient.

gioblu, 2016-04-20:
The duration can change, because of the analogRead duration, that (can be modified in the sketch), by the clock rate, the MCU and architecture. So, should be designed a universal way to accomplish this, from ATtiny to ESP. (for example ATtiny has only 3 analog input, and running at 8Mhz can be slower to compute the function).

gioblu, 2016-04-23:
Ciao @thomsna, I am testing this side. But I can't replicate this bug.
Can you please paste here the code in which id duplication occurred?

It is possible you forgot to call the receive() function in loop after id acquisition?

gioblu, 2016-04-23:
https://github.com/gioblu/PJON/commit/dd36899cc6e39f54f9c996df4973546dc82988dc

thomsna, 2016-04-24:
Hi @gioblu, sorry for the late reply. It's possible that I omitted to call the receive() function, so I've been testing the auto addressing again this afternoon, just with the example sketch.

It seems to behave incorrectly this time.
With 4 slaves, the master is receiving 2 1 1 1 2 1 1 1 2 1 1 1 ... 

Can you replicate this?

gioblu, 2016-04-25:
Ciao @thomsna I am testing will soon give you feedback.

gioblu, 2016-04-25:
Ciao @thomsna, I discovered that the random delay was not executed correctly. Thank you for your report. So I discovered the possibility of executing (obviosuly) only in the setup or loop functions.
For this reason I defined a new method: `begin()` as other protocols have. This function has to be called in the setup before any other function call to the PJON bus instance (all documentation has been updated, see readme). This finally generates quite granular delay from 0 to 1000 milliseconds. In my test setup I finally have a quite stable result.

I think in any case should be nevessary to add a cyclic sending to the ID every device owns to check for collision and in case execute back acquire_id function. For this reason I released 2.0 with `acquire_id` function considered as experimental.

gioblu, 2016-07-28:
Ciao @thomsna, how are you?
I would investigate further on this side.
You still have this sort of bug?

I can't replicate it.

thomsna, 2016-07-28:
Hi @gioblu, I won't have time to test this further the coming month, so maybe you can close this bug in the meantime, if you can't replicate it?

closed 2016-07-28 by gioblu