gioblu/PJON #58: Unable to receive data when interrups are being used.

MichMich, 2016-04-22:
Since I want to use PJON for a stepper motor project, I need to use interrups. The interrups make sure the motor runs, while the main runloop is free for other tasks (like receiving any network communication).

Unfortunately, PJON is unable to receive any messages, while the interrupts are active. If I disable the interrups, everything works fine. 

Any idea how to solve this?

I'm using https://github.com/PaulStoffregen/TimerOne to configure the interrups, and run the interrupts on a 50 microsecond interval.

gioblu, 2016-04-22:
Ciao Mich, my suggestion is to try to use interrupts on the bus, to detect the incoming packet or a initial HIGH you can generate, before transmitting, and detach the interrupts are now messing with PJON, for the time you need to receive or transmit.

gioblu, 2016-04-25:
Ciao @MichMich, 2.0 is finally out stable.
How your experiments went with the strategy proposed?
Happy tinkering :+1:

MichMich, 2016-04-27:
I've been checking out this solution, but still PJON seems to have to much of an impact on my stepper motors. I'm trying some other solutions now to see if I can get better results.

gioblu, 2016-04-29:
Maybe tihs is missing in the PJON codebase:
https://www.arduino.cc/en/Reference/NoInterrupts

Thank you for your report. I am studiyng a solution

gioblu, 2016-09-08:
Ciao @MichMich take a look to version 5!
So I would close this issue, because it is a documented known issue related to the SoftwareBitBang and OverSampling strategies. With the addition of strategies, now ThroughHardwareSerial enables to have flawless communication with those devices where interrupts are vastly used. The development of other interrupt compatible strategies will offer a wide selection of media for this sort of device.

closed 2016-09-08 by gioblu