gioblu/PJON #59: Unable to send zero.

MichMich, 2016-04-22:
I'm trying to send a byte value as a payload. Any value works, except 0. A zero won't be broadcasted (checked it with a scope.)

Works:
````cpp
network.send(1, 1, 1);
network.send(1, 2, 1);
network.send(1, 3, 1);
...
network.send(1, 255, 1);
````

Doesn't work:
````cpp
network.send(1, 0, 1);
````

gioblu, 2016-04-22:
Ciao @MichMich, yes this is an issue. if you send a single byte, the value 0 is not transmitted.
I am now near to release the 2.1, probably tomorrow, with many corrections and bugfixs, then I will dig in this direction.

aperepel, 2016-04-22:
Curious why 2.1? Did I miss the 2.0 release? The last listed for me is 2.0-beta, but I don't see the GA.

gioblu, 2016-04-22:
Ciao @aperepel ! You are of course right, 2.0 stable version
I found some other little issues but I have to test back all with all the architectures.

gioblu, 2016-04-23:
Maybe how this condition is formulated: `if(!*string) return FAIL;`
https://github.com/gioblu/PJON/blob/master/PJON.cpp#L212
Can't test now but will soon give you feedback.

sticilface, 2016-04-23:
Think you are right. The pointer is dereferenced so it checking the first character. Removing the star should fix it.

gioblu, 2016-04-23:
I tested and you are right :+1:  fixed.

closed 2016-04-23 by gioblu