gioblu/PJON #60: STM32F103R

rajdarge, 2016-04-23:
Hi there :)
Been trying to compile this library on a STMF103RB,  ARM Cortex-M3 MCU,
[http://www2.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f1-series/stm32f103/stm32f103rb.html](url)
similar to the teensy, due. Better known on the [http://www.stm32duino.com/](url)  site as the "blue pill"
unfortunately it doesn't compile. It does compile on the ESP8266 12F that I am trying to get to talk to each other. 
it complains

> 'BIT_WIDTH' ,  'READ_DELAY', 'BIT_SPACER' 'ACCEPTANCE', "was not declared in this scope"
i'm guessing that it needs an entry in the timing.h  so how to work out those values?

aperepel, 2016-04-23:
Hi @rajdarge !

Yes, if timing for the board wasn't defined, it would cause those compilation errors. This is done via a lot of trial and error, testing and tools.

For the packet length, consider that this is a very low-level chunk of data, which will get retransmitted if channel is busy, etc. It is also held in a rolling memory buffer until acknowledged by a receiving party. You probably don't want to hold too much in there.

Would building a higher-level chunking abstraction help in your case? You would have to know how to re-assemble the packets and have a buffer large enough to hold out-of-order ones.

rajdarge, 2016-04-24:
Thanks for the response:)

I'm expecting that the timing tools are beyond my level of expertise at the moment.
However there are plenty of people at STM32Duino that could be of help in this regard. I can put up a library request or you quite welcome to ask for help as there a number of people with a great deal more skill than I. Those numbers: are they at all dependent on what device is listening as well as which one is speaking? Does it need a level of hardware abstraction kinda like what you do for the mega, and even the uno?

looking at the PJON::send_string code I see you don't use null terminated data so I guess binary data CAN be sent by sending the address of the frame of data. 

As far as large packets go, I was thinking of JSON frames or  image or audio data which can be huge. I'm not yet capable of doing anything but the JSON frames, and usually I just discard the info. But WiFi creates its own problems. But when I get to that stage I'm sure I will be able to Chunk data for sending. 

A bus like this is becoming more and more important 'cause the IOT is a busy place.

zoomx, 2016-04-24:
Timings are defined in Timings.h
They are only four values 'BIT_WIDTH' , 'READ_DELAY', 'BIT_SPACER' 'ACCEPTANCE'
They depends on clock and mode.

For a initial test you can have a timings.h made of only these four values, I believe that they have to be close to the ones of ESP8266 and Arduino zero. So you can use a timings.h like this only for STANDARD mode

```
    #define BIT_WIDTH 40
    #define BIT_SPACER 112
    #define ACCEPTANCE 40
    #define READ_DELAY 4
```

gioblu, 2016-04-25:
Ciao @rajdarge.
I am happy to see that other users and contributors already gave you good indications for this porting.
How it's going?

rajdarge, 2016-04-25:
will test today. PH in aus :)

gioblu, 2016-04-25:
You still get compilation errors?

rajdarge, 2016-04-25:
yes, I'm not sure what the Compiler is calling my Board. It works fine for the esp8266

gioblu, 2016-04-25:
Clear. Which sort of framework / toolchain you are using for STM32F103R?

gioblu, 2016-04-25:
I am not sure is the toolchain you are using but, you should take a look inside the hardware directory:
i.e. https://github.com/frankpolte/stm32duino/blob/master/hardware/Arduino_STM32/STM32F4/cores/maple/boards.h#L124

rajdarge, 2016-04-25:
The hardware reference I was using was this one. I might need to post something on stm32duino asking for help. 
[STM  Hardware](https://github.com/frankpolte/stm32duino/blob/master/hardware/Arduino_STM32/STM32F1/boards.txt)

gioblu, 2016-04-25:
I don't know which board you have and which options in the boards menu but, you should look if `build.board` of your case is defined: 

i.e. https://github.com/frankpolte/stm32duino/blob/master/hardware/Arduino_STM32/STM32F1/boards.txt#L201

rajdarge, 2016-04-25:
first thing i tried :(

rajdarge, 2016-04-26:
Ok that didn't work either. But there another issue with 
PJON::initialize()
https://github.com/gioblu/PJON/blob/master/PJON.cpp#L72
references analog pin A0 which does not exist on STM32F1
[image]
These are the pinouts. for the board PB3 is the equivalent. 
I suppose something like this will have to be created to translate the pins:
```
#if defined(__SAMD21G18A__) || defined(ARDUINO_SAM_ZERO)  // Arduino Zero pins
    #define __digitalPinToPortBit(P) \
      (((P) == 0) ? PORT_PA11  : \
      (((P) == 1) ? PORT_PA10  : \
      (((P) == 2) ? PORT_PA14  : \
      (((P) == 3) ? PORT_PA09  : \
      (((P) == 4) ? PORT_PA08  : \
      (((P) == 5) ? PORT_PA15  : \
      (((P) == 6) ? PORT_PA20  : \
      (((P) == 7) ? PORT_PA21  : \
      (((P) == 8) ? PORT_PA06  : \
      (((P) == 9) ? PORT_PA07  : \
      (((P) == 10) ? PORT_PA18 : \
      (((P) == 11) ? PORT_PA16 : \
      (((P) == 12) ? PORT_PA19 : \
      (((P) == 13) ? PORT_PA17 : \
      (((P) == A0) ? PORT_PA02 : \
      (((P) == A1) ? PORT_PB08 : \
      (((P) == A2) ? PORT_PB09 : \
      (((P) == A3) ? PORT_PA04 : \
      (((P) == A4) ? PORT_PA05 : \
      (((P) == A5) ? PORT_PB02 : \
      (((P) == SCK) ? PORT_PB11  : \
      (((P) == MISO) ? PORT_PA12 : \
      (((P) == MOSI) ? PORT_PB10 : \
      (((P) == PIN_WIRE_SCL) ? PORT_PA23 : \
      (((P) == PIN_WIRE_SDA) ? PORT_PA22 : PORT_PA13)))))))))))))))))))))))))

    #define __digitalPinToPortReg(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? PORTB : PORTA )

    #define __digitalPinToPortRegOUT(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? REG_PORT_OUT1 : REG_PORT_OUT0 )

    #define __digitalPinToPortRegOUTSET(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? REG_PORT_OUTSET1 : REG_PORT_OUTSET0 )

    #define __digitalPinToPortRegOUTCLR(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? REG_PORT_OUTCLR1 : REG_PORT_OUTCLR0 )

    #define __digitalPinToPortRegPINCFG(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? REG_PORT_PINCFG1 : REG_PORT_PINCFG0 )

    #define __digitalPinToPortRegDIRSET(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? REG_PORT_DIRSET1 : REG_PORT_DIRSET0 )

    #define __digitalPinToPortRegDIRCLR(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? REG_PORT_DIRCLR1 : REG_PORT_DIRCLR0 )

    #define __digitalPinToPortINReg(P) \
      (((P) >= 15 && (P) <= 16) || ((P) == 19 ) || ((P) >= 23 && (P) <= 24) ? REG_PORT_IN1 : REG_PORT_IN0 )

    /* functions for read/write/pinmode fast for SAMD chips
     PWM pins 3, 4, 5, 6, 8, 9, 10, 11, 12, 13 provide 8-bit PWM output with the analogWrite() function
     analogWrite works on all analog pins and all digital PWM pins. You can supply it any value between 0 and 255.
     analog pins provide 12bits resolution and provide a value between 0-4096.
     * INPUT           (0x0)
     * OUTPUT          (0x1)
     * INPUT_PULLUP    (0x2) Set pin to input mode with pull-up resistor enabled
     * INPUT_PULLDOWN  (0x3) */
```

zoomx, 2016-04-26:
Use PB0, A0 was used in the original Maple Mini board and it was not used anymore because all the other boards has printed the real STM32 pin name.

gioblu, 2016-04-26:
Ciao @rajdarge, yes you are right. Pin mapping is necessary.
There we can also define A0 as PB3. If you need help I could try to write it for you also if I don't have now a chance to test it.

rajdarge, 2016-04-27:
I think I also need to change this line to 
` #if F_CPU == 72000000L`

gioblu, 2016-04-27:
Ciao @rajdarge, I passed to the thread you opened. 1-Wire GPIO strategy actually seems interesting and I am studiyng it to see if we can apply the same strategy mantaining the same support we have now. In any case, observing more closely the actual digitalWriteFast I probably started noticing parts of the mentioned complexity. Soon will be committed a refactored version.

Thank you for your valuable support to the PJON Standard.
How its going with the Maple porting?

rajdarge, 2016-04-27:
Ok, so the addition of that line allowed the timing.h file to work for my particular chip. 
Now I need a bit of help on the pin mappings:
i am using what you used here:
[digital write fast](https://github.com/gioblu/PJON/blob/master/digitalWriteFast.h#L164)
as a template beginning with:
` #if defined(__STM32F1__)`
I think this is the document that will help the most in order to translate the pins:
[STM32f1 pins](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/variants/generic_stm32f103r/board.cpp#L73)
I guess I need to know the correct syntax and if the 
> PORT_PAxx

is some kind of system variable that I should know about.

zoomx, 2016-04-28:
STM32F1 have 3 ports, A, B and C. Every control 16 pins instead of only 8 as in 8bit MCU.
But many pin have special functions
You can see a pinout here
[https://cdn.hackaday.io/images/1925191437063394286.png](url)
It is a red pill but the pins are the same as a blue pill

gioblu, 2016-04-30:
Ciao @rajdarge, I am near to push the refactored version of digitalWriteFast, still testing all the compatible architectures (quite long procedure). How it's going the porting?

gioblu, 2018-07-26:
CIao @rajdarge @zoomx @aperepel the support is now available for SoftwareBitBang, ThrouSerial and ThroughLoRa, the issue remained open I am sorry :)
If you can give it a try :)

Closing for now, feel free to re-open if required.
Thank you again for your support.

closed 2018-07-26 by gioblu