gioblu/PJON #62: Presence confimation bit

gioblu, 2016-04-29:
Ciao! Its a week I am reasoning to experiment the addition of a device presence confirmation bit after the receiver device id transmission. So the receiver should set the bus HIGH only if the packet is for its ID, giving confirmation of presence on the bus to the device that is transmitting the packet. This addition would add a bit of overhead per packet, but would extremely reduce the bandwidth occupation when a device is not receiving and another device is trying to communicate with it. So the bus would not be flooded by an entire packet transmitted multiple times, but by packet headers containing only the id, if you consider the a test case sending of a 50 bytes packet to a device is not receiving, it is an enormous optimization. This would expand the operational envelope with many devices and high bandwidth usage.

aperepel, 2016-04-29:
Such a protocol change would require versioning introduced into a handshake. Can you share what ideas you had around that? E.g. negotiate automatically, set the protocol version in begin(), etc?

gioblu, 2016-04-30:
Ciao @aperepel, this Standard optimization (if technically achievable) would provide so better results, to bring useless the use of the old version of the Standard, and in this case I would suggest the update an all devices running PJON. Probably for now and the future I would run as default the latest PJON Standard protocol version, giving the chance to the user to set through `begin` which older version of the protocol, but this bring necessary to define some sort of naming for the PJON Standard version. Any help in this field is appreciated. 

We Should before, determine the performance on media with a lot of noise like radio waves and infrared light.

I am not sure to call it handshake because we are speaking of a packet transmission and not of establishing a connection between two devices and because there are no information in the handshake procedure describing the rules of the communication.

gioblu, 2016-05-24:
I for now close this issue and consider in the future its implementation.

closed 2016-05-24 by gioblu