gioblu/PJON #70: SendAndReceive example fails with OverSampling via wireless

fredilarsen, 2016-05-14:
Using PJON 3.0 beta as of 2016-05-13.

To me it seems that sending messages in both directions like this works over a wire but not wirelessly.

Having set up two Arduino Unos with a wireless sender+receiver each, I have changed some of the examples from using SoftwareBitBang to OverSampling, and changing set_pin(12) to set_pins(6,7). The SpeedTest and NetworkAnalysis samples work perfectly over the wireless connection.

But I need something similar to the SendAndReceive example, and this will not work with OverSampling and the wireless connection. With the original example and a wire between pin 12 on each Arduino, it works.

The fact that SpeedTest and NetworkAnalysis works fine indicates that my hardware setup is fine. I can also switch which board is Device1.ino and Device2.ino and this also works perfectly.

I have attached the modified example files. The receiver is connected to pin 6 and the transmitter to pin 7 on each of the two Arduinos.

[SendAndReceive_wireless.zip](https://github.com/gioblu/PJON/files/264744/SendAndReceive_wireless.zip)

gioblu, 2016-05-15:
Ciao @fredilarsen. Thank you for your report. What sort of rf modules you are using?
I will today test it and give you feedback.

fredilarsen, 2016-05-15:
Hi, and thanks for the fast feedback.I am using cheap modules from eBay. I bought these in receiver+transmitter pairs for less than a dollar a pair.The transmitter is named FS1000A and the receiver is named XD-RF-5V. They use 433MHz and run directly on the 5V output from the Arduino board.
Regards,Fred Larsen

Date: Sat, 14 May 2016 20:13:13 -0700

fredilarsen, 2016-05-15:
Here is a picture of the modules used:
[image]

gioblu, 2016-05-15:
Ciao @fredilarsen! Thank you for the detailed report. I am testing it now.

gioblu, 2016-05-16:
Ciao @fredilarsen. Testing with a couple of wires I detect the same problem also on the blink example.
I have the sensation that:
- One of the two pins state for OverSampling strategy is working against it, creating interference.
- Some sort of eletrical interference is affecting the bus only when you send seldom, maybe pull down resistors could be required.

Further testing this afternoon.

gioblu, 2016-05-17:
Ciao @fredilarsen after extended tests I identified the lack of io pin setup for correct bidirectional communication. The bug has been fixed but I still have to test on all architectures this changes.
Thank you again for your report.

gioblu, 2016-05-17:
Will be soon released in 3.1

fredilarsen, 2016-05-17:
That is good news! I can assist in testing with Arduino Nano, Uno and Mega, with direct wire connection and with the above mentioned Wireless modules. The more testers, the merrier? :-)

gioblu, 2016-05-18:
Ciao @fredilarsen I pushed the modifications. I now am able to run the SendAndReceive example nominally. This opens up a lot of applications and seems, performance is now comparable `SIMPLEX` mode.

gioblu, 2016-05-18:
If you can test it woulld be great to validate this fix also with your hardware

fredilarsen, 2016-05-20:
I changed all included tests to the OverSampling strategy, and all tests now run successfully on the Uno clone with the wireless modules. Great work!

Unfortunately I am not able to run tests on my other board types at the moment.
I have been fighting with Windows 10 because the Mega now comes up as an unrecognized USB device. Installing the driver from ch341ser.zip does not help. Reinstalling the Arduino software does not help. Will report back when I am able.

gioblu, 2016-05-20:
Thank you @fredilarsen for discovering and reporting this issue,  :+1: 
Happy tinkering!!

closed 2016-05-20 by gioblu

gioblu, 2016-05-24:
Ciao @fredilarsen, I released 3.0 stable that contains the bus id implementation to enable a shared media communication with other channell less transceivers. Thanks to this rule is not only possible to share a medium with neighbors, but also network with them and enhance connectivity for free.