gioblu/PJON #71: What do you think about the new logo?

gioblu, 2016-05-20:
Ciao guys. I finally added the PJON's logo in the main readme.
What do you think about it? You like it?

aperepel, 2016-05-20:
Looks good, but be careful with gradients, they don't print well. Maybe have a b&w version of it or go for a few clear colors?

gioblu, 2016-05-20:
CIao @aperepel, its a while I was not hearing from you. How it's going?
Absolutely right! Take a look to the wiki, I added the new illustration we worked for months!! Hope you like!

aperepel, 2016-05-20:
Still around, and watching the PJON progress. Bit busy with work projects and Raspberry Pis recently (isn't it exciting?) :) 

I bet implementing an RPI support for PJON would be interesting, but non-RT timing nature of linux might make it difficult. But hey, I2C and SPI work on RPI, so maybe maybe...

gioblu, 2016-05-24:
Ciao @aperepel, absolutely, this is the next phase.
I now have pushed something quite extensive, if you can give it a try.

closed 2016-05-25 by gioblu