gioblu/PJON #72: I need help: serial communication

zsq19958629, 2016-06-01:
hello everyone
I really like the library.
However, it is based on single wire communication.
And my equipment works in the serial port (Uart:RX, TX) dual mode.
Excuse me, is there a way to let the library run in a two - line communication mode?
PS: my first language is Chinese, English is Machine Translation. I'm sorry!

Hardware：Arduino UNO

gioblu, 2016-06-01:
Ciao @zsq19958629, you can configure the library to use 2 pins:
https://github.com/gioblu/PJON/wiki/IO-pins-setup

And also you could write your own strategy based on Serial:
https://github.com/gioblu/PJON/wiki/Strategies

If you need support, write me back here

zsq19958629, 2016-06-01:
Thank you very much for your reply.
I'll try and hope to solve the problem.
If you can't meet the requirements I need, I'll come back and ask for your advice.
Thank you very much.

gioblu, 2016-06-01:
Sure!! Happy tinkering :+1:

gioblu, 2016-06-04:
Ciao @zsq19958629, hope your tests were positive. I close for now this issue, feel free to ask!

closed 2016-06-04 by gioblu