gioblu/PJON #74: MKR1000 compatibility

vincent-bruel, 2016-06-09:
Hi,
First of all thx for your amazing work !
I've the following error when trying to use the library with an Arduino MKR1000:
SoftwareBitBang.h:95:27: error: 'SWBB_BIT_WIDTH' was not declared in this scope

Do you plan to release/port this library for Arduino MKR1000 ?

I've tried to #if defined(__ARDUINO_MKR1000__) in Timing.h
but this is not the right define of course.
May be you know where I can find it so I can give your lib a try ?

I'm trying to communicate between MKR1000 and MEGA 2560 with a level shifter between them.
I've also tried hardware serial, but after hours it doesn't work, so I was planning to use a software library like yours on a common digital pin (1 or 2) instead.

Thanks in advance

gioblu, 2016-06-09:
Ciao @vincent-bruel. Thank you so much.
I think the main issue is:
https://github.com/gioblu/PJON/blob/master/strategies/SoftwareBitBang/Timing.h#L100

As far as I know Arduino Zero mounts the same MCU of MKR1000.
We should find where and how the indentifiyng constant is defined in the Arduino IDE codebase.

i.e. `defined(ARDUINO_SAMD_ZERO || ARDUINO_SAMD_MKR1000)`
Then should be determined if the Arduino Zero timing can be compatible with MKR1000 timing.

vincent-bruel, 2016-06-09:
Thanks you so much for your help !
I will give it a try :)

vincent-bruel, 2016-06-09:
I have also made a patch to digitalWriteFast.h (+Timing.h)

gioblu, 2016-06-10:
Good so you were able to get it working? :+1: 
Feel free to make a pull request to share your code with other PJON's users.

vincent-bruel, 2016-06-10:
Not yet, I think i've got a hardware problem (seen that on oscillo).
But the coding was easy, due to bus.set_receive(func) and filtering action on first command char, as in the examples of the lib.
It will take a few days before I can change the bogus level shifter, due to some WAF :)
Anyway I will let the community know the result.
For the pull request I've never done that, but life is nothing but a sequence of first times, so I will try to do it.

gioblu, 2016-06-13:
Good luck! and feel free to ask support!

vincent-bruel, 2016-06-14:
Hi,
I am always dealing with the hardware issue on the level shifter between the MKR1000 and the arduino MEGA.
For PJON I use 1 bus with 2 pins on each side, swapping inputs and outputs like we do with RX/TX:
At the software level with bus.set_pins(PIN_INPUT_MEGA, PIN_OUTPUT_MEGA) and but.set_pins(PIN_INPUT_MKR1000, PIN_OUTPUT_MKR1000)
At the hardware level by connecting PIN_OUTPUT_MKR1000 to PIN_INPUT_MEGA, and PIN_OUTPUT_MEGA to PIN_INPUT_MKR1000, both through my level shifter (3.3V<->5V)

On the PIN_OUTPUT_MKR1000 output line I've seen with oscillo a square wave, so the library is doing it's job: sending commands through the PJON bus.
On the PIN_INPUT_MEGA I've a constant 5.6V, and I don't know why so I need to investigate further this hardware problem.

gioblu, 2016-06-14:
Ciao @vincent-bruel, it's a pleasure to know you going forward. Yes I agree with you, seems that the library is working nominally, or at least is bit-banging out the message you coded, but for sure 5.6v on the Arduino Mega pin is not nominal. Try to use a pull-down resistor if necessary.

vincent-bruel, 2016-06-16:
Hi,
I didn't found my hardware problem and my project became really too complicated for my electronic skills, and also too complicated on the software part (that was a fact before I used your library), so I decided to start it from scratch, only with MKR1000 and 3.3V modules, and I don't need a serial library at this point.
Anyway I will recommend your library as it has a very fast learning curve, and I will keep a look at it.
Thanks for your support, and thanks for sharing.

gioblu, 2016-06-20:
Ciao @vincent-bruel, simpler is often better :+1: 
I will in any case soon buy MKR1000 and give it a try :)

Happy tinkering