gioblu/PJON #75: Analog ports as in/out pins for PJON

buzzvertising, 2016-06-13:
Hi Giovanni,

Is there an issue using the analog ports as pins for PJON?
I've put bus.set_pin(A4); on you samples and there is no way to make it work.

Do I miss something? I there a workaround?

Regards, Mircea

thomsna, 2016-06-13:
Hi @buzzvertising, I've had no problems setting A4 as a bus pin on Arduino Pro Minis.

gioblu, 2016-06-13:
Ciao @buzzvertising ! Can you tell me which hardware you are using?
On some architectures the standard timing values present in `Timing.h` can be not compatible on some pins. Bit-banging the same squarewave with different group of pins like `PORTC` or `PORTD` with a good oscilloscope can let you visually see how different they can be, doing the same with different architectures will be shocking, showing enormous differences of time perception of the microcontroller using `micros()` and `delayMicroseconds()`. PJON is designed to attenuate this problem, but on some architectures, some group of pins are so biased to generate poor communication quality or total communication inability.

buzzvertising, 2016-06-14:
Ciao @Giovanni, 
I'm using some Chinese clone of Arduino Uno. (DCcEle DCcduino UNO)
But they are installed on an an extension board and the A4, A5 pins have each a 1.5K pullup resistor. I did not consider this as a problem as I use them for I2C and they work fine.

So, I just have tried with two plain Arduino's and it works fine on analog ports:
Starting 1 second communication speed test...
Absolute com speed: 1776.00B/s
Practical bandwidth: 1480.00B/s
Packets sent: 74.00
Mistakes (error found with CRC) 0.00
Fail (no answer from receiver) 973
Busy (Channel is busy or affected by interference) 10
Accuracy: 100.00 %

Do you think that those 1.5K pullup resistors might be the problem? As transmission medium I'm using FTP cable over less than 6 m.

Thank you for your time,
Regards,
Mircea

buzzvertising, 2016-06-14:
Thank you @thomsna for your confirmation. Now it works for me too.

gioblu, 2016-06-14:
Ciao @buzzvertising, definetly, in this case timing seems not connected as I was considering.
Thank you for sharing this!

buzzvertising, 2016-06-14:
Ciao @gioblu, Solved by removing the 1.5 pull-up resistors.
Your protocol works like a charm. Thank you for a great work.
Mircea

gioblu, 2016-06-14:
It is a pleasure to know that was not useless to work 5 years on this project :+1:

gioblu, 2016-06-14:
So, I will for now close this issue, if you need support feel free to write!!
Happy tinkering :)

closed 2016-06-14 by gioblu