gioblu/PJON #76: Loopback

Kortenbach, 2016-06-18:
First of all: I really like this software ! It made me wanna redo an old project...
Today I was trying to get PJON running on my Nano.
I defined 2 separate busses, both with the same ID do do a loopback test (senderID=45 rec=44).
It didn't work, but then I read that the software does'nt use an interrupt routine, so I'm wondering:
Is it even possible to do a loopback test (can the software receive data while sending)?

Best regards,

Bert

Kortenbach, 2016-06-18:
Oh, forgot to say:
I'm using bitbang strategy on pin 9 and 10 (pins tied together)

gioblu, 2016-06-20:
Ciao, @Kortenbach I'm happy to hear you tinkering with PJON. Communication with PJON works in `SIMPLEX` and `HALF_DUPLEX`, `FULL_DUPLEX` mode is not yet supported and was never before proposed or requested as a new feature :)

PJON codebase make no use of interrupts, but I am now working to optionally enable interrupt driven watchdog for incoming packets (obviously this will not be compatible with every pin and with every architecture).

Kortenbach, 2016-06-20:
So, loopback test cannot be done?

gioblu, 2016-06-20:
As far as I know a loopback test can be done only if you transmit in full duplex.
In a half duplex system while transmitting is not possible to receive, so transmitter can't hear itself on the input line physically looping back the tx in the rx.

I will soon push some modifications that will enable the complete router functionality.
Setting a device as a router will give you the chance to receive every incoming packet in the receive function. So for example you could simply dispatch the same packet back to the sender, obviosuly this will be possibile only using PJON in network (with a defined bus id, see the network examples).

Kortenbach, 2016-06-20:
Hi Giovanni,

Thanks for your quick response! I hope I will be able to use your code, cuz
I really like it! Here is what I am trying to do:

I want to build a laser tag game in which every player sends his hitlist as
soon as he gets hit by another player. This may save me the trouble of
building a game server, but I don't want other players to miss any of these
broadcast messages just because the CPU is busy doing something else while
the broadcast was sent.

Can you tell me a bit more about the half duplex restrictions?
* If I disable the ack feedback and someone is sending me a message just
when the CPU is doing something else, will the incoming message be lost
forever? With ack enabled, the packet will be resent, right?
* With ack disabled, how long can the receiver CPU be "busy" before the
packet is lost?
* With ack enabled, how is a broadcast acked?

I didn't quite understand what you where saying in your previous mail. What
will be improved if you have changed your code? I think you may improve
quite a bit if you implement an interrupt routine for the receiver(?)

Best regards,

Bert Kortenbach

P.S. you can have my phone number if you like to chat a bit more about this
over WhatsApp...

gioblu, 2016-06-20:
Ciao @Kortenbach fantastic project!! For sure is the first time PJON is applied in such a system.

- If you disable ACK the packet could be potentially be lost forever
- If ACK is enabled the packet is resent until ACK with a cubic backoff with a maximum of 125 attempts, if the maximum is reached CONNECTION_LOST error is thrown
- With ACK disabled, to be sure not to loose neither one packet you can leave a blind timeframe of BIT_SPACER - ACCEPTANCE. Its value depends on the architecture, Strategy and communication mode. Obviously the slower you transmit, the higher is the acceptable blind timeframe.

I am now working to optionally enable interrupt driven reception, (will not work on every pin / architecture, but will cover the vast majority of use cases)

Would be a pleasure to chat, but I am, for personal choice, not using a smartphone and so whattsup, if is it ok for you, lets get in touch through email: gioscarab@gmail.com

gioblu, 2016-06-23:
CIao @Kortenbach, how its going? If you are ok with it and you don't need any other info, I would for now close this issue. Good luck and feel write to write me personally if needed

Kortenbach, 2016-06-24:
Hij Giovanni,

I am still a bit confused about your last answer. How much time can the CPU
spent on doing other stuff before it will start losing messages?

Best regards,
Bert kortenbach

gioblu, 2016-07-15:
Ciao @Kortenbach, being every byte predended by padding bits or a header, the maximum time dedicated not to reception should not exceed `BIT_SPACER` - `ACCEPTANCE` (see  https://github.com/gioblu/PJON/wiki). How much time is this value depends on which PJON strategy, architecture and clock, see the Timing.h file of the strategy you use to find the values you need.

(I am sorry for the late answer to your question)

closed 2016-07-28 by gioblu