gioblu/PJON #83: PJON with only two wires

phpfs, 2016-08-08:
Hello!

Can PJON be used with only two wires in total which also send the energy for the arduino?

Example:
--------- Ground
--------- 5V and Data

Thanks you :)

gioblu, 2016-08-08:
Ciao @phpfs parasitic power mode (like 1-Wire) is still not an available functionality.
What could be done to enable this over PJON is develop a strategy able to use 1-Wire as communication medium.

gioblu, 2016-08-08:
@phpfs or obviously try to develop a parasitic version of SoftwareBitBang, but it is quite a work

drtrigon, 2018-06-11:
I would also like to see that feature, can we re-open the issue?

gioblu, 2018-06-11:
Sure @drtrigon  :)
You would need to have parasitic power mode on PJDL or you would have data-link 1-Wire support?

drtrigon, 2018-08-13:
I think we would need to look at several things:
* we need to add a capacitor from data line to gnd for every device; what is the optimal value? (might heavily depend on used hardware)
* SWBB need to be in high state when busy and timing needs to have plenty of busy time
* investigate on compatible hardware and needed settings (power budget, brown-out detect, watchdog, etc.)
* most of the code might need to be run in `setup()` instead of `loop()` procedure (atmega)
* use of EEPROM for variables or states could be useful

drtrigon, 2018-08-17:
> There is no way to get it working in multi-master for obvious reasons (the line must be kept high by master and request must be sent by master to devices to obtain responses)

No I don't agree on this:
* the bus gets a pull-up resistor to be kept high (in idle state) all devices (any number of masters and slaves can charge - may be the value of the pull-up needs to be adopted for big number of devices)
* HIGH state is passive by setting the pin to input (high impedance)
* LOW state is active by pulling the pin low (low impedance)
* the total time of low states should be small

This way the line is kept high independent of any master devices. Multi-master and with this SWBB will be possible, or do I miss anything?

gioblu, 2018-08-18:
Ciao @drtrigon thank you, I am not sure using with the internal pull-up resistor you can get enough current to power all connected devices. For sure it is easier to use master-slave approach in this particular case, although it may be possible to get it working as you describe, probably using an external pull-up.