gioblu/PJON #89: Porting PJON to C (To support limited devices)

Matheus-Garbelini, 2016-09-03:
Hello. Are there any planning in doing a lightweight version of this library?. Currently i'm porting to C only SoftwarerBitBang startegy.

gioblu, 2016-09-04:
CIao @Matheus-Garbelini happy to hear from you, would be amazing to have PJON available also in C.
I would ask you as I asked to Girgitt, that is developing the python version, to mantain the same sort of license, and obviously acknowledge the PJON Standard copyright in your files.
Feel free to write me personally for support.
Take a look to the repo because soon will be released 5.0-beta!

gioblu, 2016-09-05:
Ciao @Matheus-Garbelini Take a look to the official specification of the Padded jittering da link implimenented by the SoftwareBitBang and OverSampling strategies:
https://github.com/gioblu/PJON/blob/master/strategies/SoftwareBitBang/specification/padded-jittering-protocol-specification-v0.1.md

Here the specification for the PJON protocol layer:
https://github.com/gioblu/PJON/blob/master/specification/PJON-protocol-specification-v0.3.md

gioblu, 2016-09-16:
Ciao @Matheus-Garbelini how its going?

Matheus-Garbelini, 2016-09-20:
Sorry for the delay. I managed to get it working, but it's in a older version (4.2), and i'm thinking in updating it and running more tests on it before posting it completly with examples included, code cleaned up and such on.
As i'm working on it in parallel with the project i'm doing, the porting is being naturally implemented.

If you have interest in looking how it is right now, you can give a look in:
[https://github.com/Matheus-Garbelini/PJON_C](url)

gioblu, 2016-09-20:
Ciao @Matheus-Garbelini I am reading your work. 
Fantastic! It is an honour to see PJON implemented in C. :) 

Please feel free to ask for all the support you need to keep it compliant and interoperable with the other compliant tools. 

If I can give you a suggestion I would consider the use of a constant to define which strategy you want, something like #define SOFTWARE_BIT_BANG true

if SOFTWARE_BIT_BANG
- definition of functions of the related strategy
- definition of timing values accordingly

this could enable the easy addition by other contributors of other strategies! 

Because of how you structured it, it is impossible to run more than one PJON instance, also because the instance doesn't exist :). But for small devices could be really nice and slick! Have you already compared the memory footprint? Have you tried to run this version transmitting or receiving with device programmed with the PJON C++ implementation?

gioblu, 2016-09-20:
@Matheus-Garbelini consider that your implementation should:
- be licensed as PJON is so with Apache 2.0 
- If you can, mantain a sort of Standardization and so follow the project structure of PJON, see PJON-python: https://github.com/Girgitt/PJON-python
- Remember to include the acknowledgment
- Include specification (see PJON-python)

Would be nice if is ok for you to switch to PJON-C instead of PJON_C.
Because we are not working on PJON, PJON-JS, PJON-PHP and girgitt is working on PJON-python. So would be nice to be aligned!

Matheus-Garbelini, 2016-09-20:
Thank you, i will update it with the corrections. Also, I tested it with, with the c++ implementations and it works fine, c++ and c implementations works together. 

So, my idea was to allow just one instance for testing purpoises, As it is now, the memory footprint is around 3kB in a 32bits system with optimize for size option checked.

When i finish the porting, i will cleanup it and include the correct specification for it.
Thank you a lot to PJON, it is a huge and powerfull protocol xD

jaapcammeraat, 2017-06-02:
Where can I find the PJON-C  implementation?

gioblu, 2017-06-04:
ciao @jaapcammeraat there is no updated and compliant implementation in pure C, @Matheus-Garbelini have started some time ago working on it, but it seems now obsolete and not maintained: https://github.com/Matheus-Garbelini/PJON-C