gioblu/PJON #90: Interrupts messing with the message content?

buzzvertising, 2016-09-05:
Hi Giovanni,
I'm using your library to make a series of devices communicate.
Here is the code for two of them:
https://codeshare.io/otiat
https://codeshare.io/cNdWX

The problem is that both id and state value I'm sending gets random values on the receiver end.
Is this still an issue with the interrupts or I am missing something.

Best Regards,
Mircea

gioblu, 2016-09-05:
Ciao @buzzvertising happy to hear from you. The best test to determine if the problem is the interrupt interference issue is to run the same sketch with the ThoughHardwareSerial strategy that uses interrupts and should not be affected by them leveraging of a different detection procedure!

Which version of the repo you are using?

buzzvertising, 2016-09-06:
Ciao Giovanni,
I'm using PJON v4.1 as latest repository was reporting some errors on compiling. Was easier to  downgrade to get rid of errors :)
I'll follow your advice update to latest PJON and use ThoughHardwareSerial. I'll come back with a report.
Thank you for your support and time.
Regards,
Mircea

gioblu, 2016-09-06:
Ciao @buzzvertising  Its a pleasure to help, let me know your results

gioblu, 2016-09-15:
Ciao @buzzvertising have you got some results? Need some support?

closed 2016-11-07 by gioblu