gioblu/PJON #92: Sudden comm problem

dvv, 2016-09-13:
Hi!

Do not know if it is PJON's issue but still if I add
```C
Serial.print("Temperature "); Serial.println(temperature, DEC);
```

right after https://github.com/gioblu/PJON/blob/master/examples/Local/SendArbitraryValues/Transmitter/Transmitter.ino#L28 the communication stops:

```
Temperature 1023
Temperature 1023
Connection lost with device id 44
Temperature 1023
Connection lost with device id 44
Temperature 1023
Connection lost with device id 44
Temperature 1023
```

Wonder how to fix that?

TIA,
--Vladimir

gioblu, 2016-09-13:
Ciao @dvv could you please tell me the version of PJON, which board you are using for communication (both) and which strategy you are using?

dvv, 2016-09-13:
5.0.0 via arduino lib manager, both nano328p16MHz.
Oh, forgot to mention I ran tests in proteus.
They work until I add the mentioned line. And if I comment out either of prints, it still works.

gioblu, 2016-09-13:
Ciao @dvv the problem is the example itself :) and in this case, my fault.
This should work fine:
```cpp
#include <PJON.h>
int packet;
// <Strategy name> bus(selected device id)
PJON<SoftwareBitBang> bus(45);

unsigned long time = millis();

void setup() {
  bus.strategy.set_pin(12);
  bus.begin();
  bus.set_error(error_handler);
  Serial.begin(115200);
  int temperature = analogRead(A0);
   //Send a "T", break the int in two bytes
  char content[3] = {'T', temperature >> 8, temperature & 0xFF};
  packet = bus.send(44, content, 3);
}

void error_handler(uint8_t code, uint8_t data) {
  if(code == CONNECTION_LOST) {
    Serial.print("Connection lost with device id ");
    Serial.println(data);
  }
}

void loop() {
  if(!bus.packets[packet].state) {
    temperature = analogRead(A0);
    //Send a "T", break the int in two bytes
    char content[3] = {'T', temperature >> 8, temperature & 0xFF};
    packet = bus.send(44, content, 3);
  }
  bus.update();
};
```
 So I am saving the pointer to the packet in the setup and I use it in the loop to detect if was delivered and if was schedule it to be sent again! The method send, includes a packet in the buffer and will try to send it at the first update call. 

I will soon publish a fix to this reported inconsistency / bug in this example.
Thank you for your support to the PJON project.

dvv, 2016-09-13:
Yes, `if(!bus.packets[packet].state) {` seems a bit of snooping.
Thank you! I'll wait for the example updated.

gioblu, 2016-09-13:
You can also call `send_repeatedly(id, content, length, timing)`
i.e. `bus.send(44, content, 3, 1000000);` to send the packet every second.

Soon update release.

gioblu, 2016-09-14:
Ciao @dvv I have made some tests. As always I take for granted to have made a mistake. 
Seems the proposed sketch is working nominally, and the packet is sent once per second as it should, the snooping is obviously not necessary and the packet sending procedure is ok (so I erase a part of my answer not to mislead users). 

Analizing better the data you submitted seems transmitter is affected by, potentially, Serial transmission interrupt driven procedure, creating interference/distortion in the software bit banging procedure, but it is only a guess. So absolutely PJON related, thanks for reporting,

gioblu, 2016-09-14:
Ciao @dvv Could you please run this test again with the addition of a pull down resistor on the bus line?

dvv, 2016-09-14:
Mmm, I believe there're no noise in lines in proteus. But I'll do, sure.

gioblu, 2016-09-14:
Ciao @dvv I missed the fact you are simulating on Proteus.
I never tried to simulate with it, and I am sincerely not sure if is it able to be coherent at microsecond resolution with the bits produced by a single task non concurrent program. Instead Proteus runs in a system where there are multpiple threads running together at the same time, this could lead to malfunction. There is also the chance of execution time simulation accuracy, that is stricly architecture dependant.

gioblu, 2016-09-14:
@dvv try switching Strategy from SoftwareBitBang to ThroughHardwareSerial

dvv, 2016-09-14:
Well, time accuracy I believe to be _the same for all_ chips involved (both tx and rx), hence I doubt the second assumption.

gioblu, 2016-09-14:
Ciao @dvv Every architecture produces a slightly different trace if looked with an oscilloscope (with SoftwareBitBang and OverSampling). This because every architecture executes the PJON codebase with a different pace because of their differences (clock, processor, other software running i.e. an operative system). Infact, as you can see in the TIming.h file, the timing values are different on architectures to generate bits as similar as necessary to have 100% accuracy. What I meant that I am dubious that proteus takes into account those small differences.

dvv, 2016-09-14:
What I do know that it simulates quite well bitbanged 1-wire and I believe 1-wire's timings are quite tight enough.
Again, I do not insist this is a PJON issue and have said in my very first sentense.
Wonder if someone else could as well confirm/void simulation suspection.

gioblu, 2016-09-14:
Ciao @dvv I will make some tests with it tonight. 
Thanks for sharing.

gioblu, 2016-09-15:
Ciao @dvv see https://github.com/gioblu/PJON/commit/be9d7ed9d5e5f4f8665b67b3063bc9b11dec98f1

dvv, 2016-09-20:
I've tested _the master_ with a pair of tranceivers (ids switched):

```c
//------------------------------------------------------------------------------

#include <PJON.h>

PJON<SoftwareBitBang> bus(44);

//------------------------------------------------------------------------------

void receiver_function(uint8_t *payload, uint8_t length, const PacketInfo &packet_info)
{
  Serial.print("RX:");
  Serial.print(" Device id: ");
  Serial.print(packet_info.receiver_id);
  Serial.print(" | TX: ");
  Serial.print(" Device id: ");
  Serial.print(packet_info.sender_id);

  if((char)payload[0] == 'V') {
    Serial.print(" | Voltage: ");
    Serial.print(payload[1] << 8 | payload[2] & 0xFF);
    Serial.print("mV");
    // Get back the original integer from the 2 separate bytes
  }
  Serial.println();
}

//------------------------------------------------------------------------------

void setup()
{
  bus.strategy.set_pin(2);
  bus.begin();
  bus.set_receiver(receiver_function);
  Serial.begin(38400);
  Serial.print("PJON - Device id ");
  Serial.print(bus.device_id());
  Serial.println(" A0 pin voltage cyclical packet sending...");
}

//------------------------------------------------------------------------------

unsigned long time = millis();

void loop()
{
  if (millis() - time > 750) {
    static int packet;
    time = millis();
    int voltage = (float)(5.0 / (1023.0 / analogRead(A0))) * 1000;
    Serial.print("Voltage: ");
    Serial.print(voltage, DEC);
    Serial.print("mV ");
    /* Avoid Serial and PJON concurrency */
    Serial.flush();

    /* Send a "V", break the int in two bytes */
    char content[3] = {'V', voltage >> 8, voltage & 0xFF};

    unsigned long send_time = micros();
    /* Use a blocking version of send. */
    packet = bus.send_packet_blocking(45, content, 3);

    /* Determine communication result and duration */
    send_time = micros() - send_time;
    float duration = send_time / 1000.0;
    Serial.print((packet == ACK) ? " Transmitted" : " Some error occurred");
    Serial.print(" - Duration: ");
    Serial.print(duration);
    Serial.print(" milliseconds");
    Serial.print(" - Communication speed: ");
    //               1 second / (duration / (overhead + length + ACK))
    Serial.print((int)(1000.0 / (duration / (bus.packet_overhead() + 3 + 1))));
    Serial.println("B/s");
  }
}
```

The comm not performed -- https://gyazo.com/1e7980b6e57fa12243a11305aa65796f.
I noticed the line stays high after the first burst -- is it expected?

gioblu, 2016-09-21:
Ciao @dvv I am happy to see you digging.
I dont see how the line can remain up after transmission:
- The last bit of ACK is 0
- The first action of receive_byte is to pull down the line https://github.com/gioblu/PJON/blob/master/strategies/SoftwareBitBang/SoftwareBitBang.h#L101

So in what ever case, on both sides the line should end up down after transmission.

I am now testing with a couple of devices, Arduino duemilanove and Arduino Mega:
```cpp
Voltage: 518mV  Transmitted - Duration: 5.34 milliseconds - Communication speed: 1686B/s
Voltage: 518mV  Transmitted - Duration: 5.34 milliseconds - Communication speed: 1686B/s
Voltage: 518mV  Transmitted - Duration: 5.33 milliseconds - Communication speed: 1689B/s
Voltage: 713mV  Transmitted - Duration: 5.33 milliseconds - Communication speed: 1689B/s
Voltage: 5000mV  Transmitted - Duration: 5.32 milliseconds - Communication speed: 1690B/s
Voltage: 5000mV  Transmitted - Duration: 5.32 milliseconds - Communication speed: 1690B/s
```
At rest and connected to the 5v pin. If I simply plug the data pin to led 13 I can see it blinking progressively slower until shut down (the cubic backoff in action and the visual demonstration that between transmissions the bus is pulled low).

Are you running custom configuration or with the default one?

gioblu, 2016-10-13:
Ciao @dvv, could you give a try to it with a different strategy to see if the problem occurs also in that case, could help to clear out this issue

dvv, 2016-10-28:
Ciao @gioblu.

Returned to the problem. Used the current master. First node transmits once a 1000ms, second once a 750ms.
https://gyazo.com/ad8ab15110cb2e547a1d70e2e66adcfc shows there's a clinch after 2 comms (line is highly active, but error -1 reported), and after it is somehow resolved, comm continues without error (as far as I see for several minutes).

I thought it can be linked to both A0 (used as source of randomness at the beginning) were the same in sumulation and pulled them up and down https://gyazo.com/f67e8aecf2e6b9e5e3713389b04e2d97. No changes.

Changed device numbers (which might be used as additional level of randomizing backoff), no changes.

---

How to wire devices for hardware serial?

gioblu, 2016-10-28:
Yes the A0 issue can be the solution.
Are you printing the result of send there? If so, being packet an int (not unsigned), maybe it prints -1 instead of 65535 (FAIL)

gioblu, 2016-10-28:
@dvv probably I should change the type in every examples to avoid this issue if possible

dvv, 2016-10-28:
`int packet`, right.
As I said, I pulled A0 of two devices to ground/power respectively, and saw no changes in behavior.
But look, relying on A0 being _float_ is a failure for some setups (say, those which use A0 as normal input).
Couldn't you fork a testing branch on which you would add additional randomness based on device ID so I could test it?
TIA

gioblu, 2016-10-30:
ciao @dvv I am sorry for the late answer.
I agree with you that would be interesting to use the device id if present.
Do you know if randomSeed can be called more than once? Because if so could be set again after dynamic addressing acquisition. I dont understand your point about float.
I can fork for you tomorrow! Thank you again for your support to the project.

dvv, 2016-10-30:
I mean if one has schematic where A0 pulled down at startup, no randomness is introduced, since `randomSeed(0)` should result in constant seed. No?

gioblu, 2016-11-01:
Ciao @dvv you are right, I didn't consider this possibility.
So you suggest to call randomSeed passing the device_id if is defined, or call again randomSeed, after id acquisition passing the new id?

dvv, 2016-11-02:
Well, I though `device_id` is hardcoded. Something like that. Or leave option to define the pin for source of randomness.

gioblu, 2016-11-02:
Ciao @dvv `device_id` can be hardcoded, but if you instance PJON<> bus without parameters, your device is initialized with `NOT_ASSIGNED` id and dynamic addressing procedure will start after begin.

gioblu, 2016-11-02:
Ok @dvv , so I will add setter for randomness pin, and use device_id if present for randomness. I hope this will fix this issue.

gioblu, 2016-12-03:
Ciao @dvv here is the commit you asked for, sorry for being late https://github.com/gioblu/PJON/commit/27ac42744f860b857388f7e2c2cc0f2195a74b46

gioblu, 2016-12-11:
Ciao @dvv trying to revive this issue after version change again. 6.1 seems have partially fixed some of the proteus errors also if I have obtained few other reports by users, have you tried 6.1 in the Proteus environment?

dvv, 2016-12-15:
Ciao @gioblu much better now. Thank you!

gioblu, 2016-12-15:
Ciao @dvv thank you for your feedback, can I consider this issue closed?

dvv, 2016-12-15:
I think yes.

closed 2016-12-15 by dvv