gioblu/PJON #93: Is possible to use two strategies in one sketch?

Radiusxe, 2016-09-16:
Hi,
is possible to use two strategies in one sketch? I need to send data over SoftwareBitBang (Arduino A to ESP) and to use OverSampling to receive data (Arduino C to Arduino A).

Currently I don't know, how to make second instance of PJON, it is not possible right? Is there some kind of solution?

Very thanks Radius

gioblu, 2016-09-17:
ciao @Radiusxe Absolutely, possible :)

```cpp
PJON<SoftwareBitBang> wired_bus(10)
PJON<OverSampling>    radio_bus(10)

void setup() {
  wired_bus.begin();
  radio_bus.begin();
}

void loop() {
  wired_bus.receive();
  radio_bus.receive();
}
```

gioblu, 2016-09-17:
And so beautiful things like 
```cpp
wired_bus.set_router(true);
wired_bus.set_receiver(wired_receiver_function);

void wired_receiver_function(uint8_t *payload, uint8_t length, const PacketInfo &packet_info) {
  radiio_bus.send(packet_info.receiver_id, payload, length); 
}
```

So every packet travelling in the wire will be mirrored on the radio

Radiusxe, 2016-09-17:
Hi,
great very thanks.

Radius

closed 2016-09-17 by Radiusxe