gioblu/PJON #94: Dynamic Addressing

Darksecond, 2016-09-22:
I was wondering if there is a way to dynamically assign addresses with PJON, instead of statically assigning them.

gioblu, 2016-09-22:
Ciao @Darksecond, It is a feature scheduled to be added soon! :), we are now working on master-slave auto addressing, with the addition of a new class to be used on the master and some changes on the original PJON class.

Matheus-Garbelini, 2016-09-27:
This is awesome. Currently i have to use series switches (relays) to auto address slaves by their relative position to the master: The bus pass in series by each slave relay, whose can cut the communication to the next slave. When the circuit power up. All slaves cut the next line, but when the master configures the first, the slave open the line to the second and so on... just an idea.

gioblu, 2016-09-29:
Ciao @Matheus-Garbelini How is the C porting going?
Speaking about this issue, we are working on a master-slave with multi-master fallback in absence of master. Using a new class `PJONMaster` that inherits `PJON` and adds the master functionality. On the other side in `PJON` has been added the slave functionality necessary for the exchange:

```cpp
#define ID_REQUEST  200
#define ID_CONFIRM  201
#define ID_NEGATE   202
```
`rid` = random generated 4 bytes unique identifier

1. Slave with `NOT_ASSIGNED` id asks for a new device id sending `ID_REQUEST` and its rid, master receiving this packet will reserve an id for the rid.
2. Master answers with a `BROADCAST` containing `ID_REQUEST`, rid and the new device id reserved
3. Slave with new device id asks for confirmation sending `ID_CONFIRM`, rid and device id
4. Master ends broadcasting, checks for rid collision, if detected sends `ID_NEGATE` forcing the slave to generate a new rid and restart the process. if not the procedure is finished.

In the multimaster case:
1. Device extract a random number and tries to contact that device
2. If device answers, it adds one to the id and tries again
3. If any answer is obtain from a certain id that is taken, the device stays for a random time in reception (to be able to answer to other devices potentially interested in that id) 
4. The device tries to contact itself to probe potential collision, if any answer is received the device is considered taken.

Matheus-Garbelini, 2016-09-29:
Hello gioblu, still working on it and doing more test. i Believe it will be nice next month.

Thank you a lot by this feature, it looks very great. Also could you also add an optative callback that masters or slave execute on auto addressing success? It would be useful to make decisions on every new registration, possibly enabling interaction with an external peripheral that could be also interacting with the bus.

gioblu, 2016-10-02:
@Darksecond, @Matheus-Garbelini, see https://github.com/gioblu/PJON/blob/master/specification/PJON-dynamic-addressing-specification-v0.1.md

gioblu, 2016-10-02:
Ciao @Darksecond @Matheus-Garbelini  here is the example:
https://github.com/gioblu/PJON/tree/master/examples/Network/MasterSlaveAutoAddressing

Here the file: https://github.com/gioblu/PJON/blob/master/PJONMaster.h

Please if you can give it a try, I would be happy to know your opinion and test results.

For now the Master is able to get a device disconnected only the first time it will try to speak with it.
There is no continuous check not to consume bandwidth.

Maybe the slave part of should be extracted in a PJONSlave class?

gioblu, 2016-10-02:
@Darksecond @Matheus-Garbelini extracted in PJONSlave: https://github.com/gioblu/PJON/commit/a667d8860398c2cf95c602229b82a1498abfa4ea

gioblu, 2016-10-15:
Ciao @Darksecond @Matheus-Garbelini have you got any test of this new feature?

Darksecond, 2016-10-17:
haven't gotten the chance, sorry!

Matheus-Garbelini, 2016-10-18:
Sorry for the delay. Planning to test next week!!

elusive-code, 2016-12-06:
Hi, I just found this project, and it looks really interesting. 
However it seems that example mentioned doesn't work when there are several slaves.

I tried simple setup with one master (Arduino Mega2560) and 2 slaves (Arduino Uno and Arduino Mega2560).

Master outputs in serial only one device. 
From what I can tell - the one that was first to connect (I tried connecting only one slave at a time - both worked).

gioblu, 2016-12-06:
Ciao @elusive-code thank you for your feedback, will run again tests tonight and dig out!

gioblu, 2016-12-07:
Ciao @elusive-code it is strange, I made a test with an Arduino Duemilanove an Arduino Mega used as a master and an Arduino Micro and it works quite flawlessly printing:
```
Header: 10111 Length: 5 Sender id: 255 Request: 200 RID: 4294939499 Packet: 200 255 255 147 107 Packets in buffer: 1

Header: 10111 Length: 6 Sender id: 1 Request: 201 RID: 4294939499 Packet: 201 255 255 147 107 1 Packets in buffer: 0

Header: 10111 Length: 5 Sender id: 255 Request: 200 RID: 4294937722 Packet: 200 255 255 140 122 Packets in buffer: 1

Header: 10111 Length: 6 Sender id: 2 Request: 201 RID: 4294937722 Packet: 201 255 255 140 122 2 Packets in buffer: 0

Device id: 1 Device rid: 4294939499 | Device id: 2 Device rid: 4294937722 | 
```
Which strategy you are using?

elusive-code, 2016-12-08:
@gioblu, I tried running code in the example - it uses SoftwareBitBang strategy. Only modification I made is pin used (I changed 12 to 2).

Maybe I did something wrong in the setup? I just connected them directly in one line through breadboard. Does it require pull-up (or pull down) resistor? 
(Actually I think I tried with pull-up resistor too, because before I was checking out 1-wire, though I'm not sure)

gioblu, 2016-12-08:
CIao @elusive-code, pull down resistor could be necessary in some use cases / depending on you grounding / power supply. Do not forget share a common ground for all devices

elusive-code, 2016-12-08:
Ahhh, yes, the common ground, it seemed that was the issue, now it works, thanks!

Btw I noticed that method acquire_id_multi_master is in PJONSlave, does it mean that for multimaster mode PJONSlave should be used? 
Is it possible then add ids list to PJONSlave as well (to have information on who is online)?

Also, while speaking of multimaster, is there some kind of bus arbitration? like when two devices trying to write at the same time? (couldn't find that in specs)

And another question:
Is hot swap supported or planned for support? Or is there a way to support it from "userspace", like checking periodically in loop?
I tried to unplugging (the data pin I mean, not power) device - and master still listed it.
When I rebooted slave arduino it connected as new device (master now sees three of them, but physically only two are there):
```
Device id: 1 Device rid: 4294935689 | Device id: 2 Device rid: 4294950752 |
Device id: 1 Device rid: 4294935689 | Device id: 2 Device rid: 4294950752 |
Header: 10111 Length: 5 Sender id: 255 Request: 200 RID: 4294948428 Packet: 200 255 255 182 76 Packets in buffer: 1
Header: 10111 Length: 6 Sender id: 3 Request: 201 RID: 4294948428 Packet: 201 255 255 182 76 3 Packets in buffer: 0
Device id: 1 Device rid: 4294935689 | Device id: 2 Device rid: 4294950752 | Device id: 3 Device rid: 4294948428 |
Device id: 1 Device rid: 4294935689 | Device id: 2 Device rid: 4294950752 | Device id: 3 Device rid: 4294948428 |
```

And one more thing: 
while it wasn't working I noticed that slaves printed in serial it's device_ids (not 255 - NOT_ASSIGNED, but actual ids), even though master wasn't printing both of them.
I guess it kind of implies that slaves thought that their ids were confirmed, even though they were not.
Not sure if that is something to be concerned about, considering the cause, or even if it is possible to reproduce to debug.

gioblu, 2016-12-08:
@elusive-code  PJONMaster and PJONSlave are abstractions over PJON base class (that includes multiplexing and traffic control features, to answer you to the arbitration question). You could use both multi-master an master-slave with the PJON base class implementing on top the procedure.
PJONSlave and PJONMaster are offering the addressing procedure. The slave contacts master to have an id, if master not present there is a fallback to acquire_id_multimaster.

The master can't be aware that a device is not present anymore if there is no communication to it ongoing. I didn't implement a continuous polling mechanism because it could be heavy on bandwidth. The messages the user will send and receive in case of failure to a device on master side, will trigger the id deletion in case of CONNECTION_LOST error, but if you have a better idea, feel free to share! :)

About the last question, the id you see printed is the one obtained by the device thanks to acquire_id_multimaster.

elusive-code, 2016-12-08:
Thanks! 
Auto-removal on failed delivery is actually a good solution for me. I can just send heartbeat messages if need be.

I believe I found an actual issue related to splitting PJON into PJONMaster and PJONSlave #106

gioblu, 2016-12-11:
Ciao @elusive-code thank you for your support exposing the issue 106, I fixed thanks to your suggestions.

gioblu, 2016-12-22:
@elusive-code @Darksecond @Matheus-Garbelini The feature is now considered stable after extended testing. I will close this issue, feel free to reopen it in case of bug report or change proposal

closed 2016-12-22 by gioblu