gioblu/PJON #96: ThroughSerial and ESP8266

fastriker, 2016-10-05:
Hi, 
I get the following error message when I include the PJON library  and want to compile for an ESP8266 board.
```
/home/user/Arduino/libraries/PJON/strategies/ThroughSerial/ThroughSerial.h:27:28: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
```

My minimum example:
```
#include <PJON.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```
Regards,
Kai

gioblu, 2016-10-05:
Ciao @fastriker. Recently we added the compatibility for SoftwareSerial, that is included in the top of the file. To fastly fix the problem on your side, add in the libraries directory an implementation of SoftwareSerial for you architecture, i.e. : https://github.com/plerup/espsoftwareserial

I will evaluate how to avoid this for the ESP user, thank you for pointing out.

gioblu, 2016-10-06:
The problem should be fixed with: https://github.com/gioblu/PJON/commit/074612f931eada6fa0c987424555f388ea9692a6

Is duty of the user to include and declare the sort of Serial he needs :) 
I don't know why I forgot there a static inclusion.

Thank you again.

fastriker, 2016-10-06:
Nice :) No matter

closed 2016-10-06 by fastriker