gioblu/PJON #98: Error compiling: 'SoftwareSerial' has not been declared

Elbas, 2016-10-06:
Hello,

I am experimenting with making a wireless network with RF modules on my Arduino for home automation, and stumbled upon this library. As far as I can see for now, its exactly what I am looking for. Great work!

But.... When I try the example on the wiki ( https://github.com/gioblu/PJON/wiki ), I get the following error message:
```
Arduino: 1.6.10 (Linux), Board: "Arduino/Genuino Uno"

/home/bas/Downloads/arduino ide/arduino-1.6.10/arduino-builder -dump-prefs -logger=machine -hardware "/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware" -tools "/home/bas/Downloads/arduino ide/arduino-1.6.10/tools-builder" -tools "/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/tools/avr" -built-in-libraries "/home/bas/Downloads/arduino ide/arduino-1.6.10/libraries" -libraries "/home/bas/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10610 -build-path "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/bas/Arduino/pjon_zender_test/pjon_zender_test.ino"
/home/bas/Downloads/arduino ide/arduino-1.6.10/arduino-builder -compile -logger=machine -hardware "/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware" -tools "/home/bas/Downloads/arduino ide/arduino-1.6.10/tools-builder" -tools "/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/tools/avr" -built-in-libraries "/home/bas/Downloads/arduino ide/arduino-1.6.10/libraries" -libraries "/home/bas/Arduino/libraries" -fqbn=arduino:avr:uno -ide-version=10610 -build-path "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/home/bas/Arduino/pjon_zender_test/pjon_zender_test.ino"
Using board 'uno' from platform in folder: /home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr
Detecting libraries used...
"/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/variants/standard" "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/sketch/pjon_zender_test.ino.cpp" -o "/dev/null"
"/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/variants/standard" "-I/home/bas/Arduino/libraries/PJON" "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/sketch/pjon_zender_test.ino.cpp" -o "/dev/null"
"/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/variants/standard" "-I/home/bas/Arduino/libraries/PJON" "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/sketch/pjon_zender_test.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/variants/standard" "-I/home/bas/Arduino/libraries/PJON" "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/sketch/pjon_zender_test.ino.cpp" -o "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/bas/Downloads/arduino ide/arduino-1.6.10/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10610 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/cores/arduino" "-I/home/bas/Downloads/arduino ide/arduino-1.6.10/hardware/arduino/avr/variants/standard" "-I/home/bas/Arduino/libraries/PJON" "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/sketch/pjon_zender_test.ino.cpp" -o "/tmp/buildc47d0769f58d5076b3177e3e96f2a238.tmp/sketch/pjon_zender_test.ino.cpp.o"
In file included from /home/bas/Arduino/libraries/PJON/PJON.h:76:0,
                 from /home/bas/Arduino/pjon_zender_test/pjon_zender_test.ino:1:
/home/bas/Arduino/libraries/PJON/strategies/ThroughSerial/ThroughSerial.h:109:21: error: 'SoftwareSerial' has not been declared
     void set_serial(SoftwareSerial *serial_port) {
                     ^
/home/bas/Arduino/libraries/PJON/strategies/ThroughSerial/ThroughSerial.h: In member function 'void ThroughSerial::set_serial(int*)':
/home/bas/Arduino/libraries/PJON/strategies/ThroughSerial/ThroughSerial.h:110:14: error: cannot convert 'int*' to 'Stream*' in assignment
       serial = serial_port;
              ^
Using library PJON at version 5.2 in folder: /home/bas/Arduino/libraries/PJON 
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

It can be some sort of stupid mistake or misunderstanding from my side. Have not really investigated this yet, but I thought I'll let you know.

Is this known?
How to fix this?

Thank you for the great work!

gioblu, 2016-10-06:
Ciao @Elbas the linked commit above should fix this issue.
Generalizing the setter parameter type to `Stream` should avoid this error.

Thank you so much for pointing out, and for your compliments :)
Try to pull master version and give it a try!

Elbas, 2016-10-08:
Changed the lines in my local library. That solved the issue.
Data transfer by wire is going as expected.
Thank you!

Next issue to tackle is the range of the RF modules (like 50cm/20inch max on quick experiment), but thats not a topic to discuss here. 

Keep up the good work!

closed 2016-10-08 by gioblu